loongson/Firmware #1: A2101 - hard-bricked if Above 4G Decoding is enabled

xen0n, 2022-06-17:
```
MemAbove4G.Base 0x80000000
MemAbove4G.Limit 0xFD00000000
Io.Base 0x20000
Io.Limit 0x2020000
Mem.Base 0x20000000
Mem.Limit 0x80000000
RootBridge: PciRoot(0x0)
  Support/Attr: 7007F / 7007F
    DmaAbove4G: Yes
NoExtConfSpace: No
     AllocAttr: 3 (CombineMemPMem Mem64Decode)
           Bus: 0 - FF
            Io: 20000 - 2020000
           Mem: 20000000 - 80000000
    MemAbove4G: 80000000 - FD00000000
          PMem: 0 - 0
   PMemAbove4G: 0 - 0
PciHostBridge driver failed to set EFI_MEMORY_UC to MMIO aperture - Unsupported.
PciHostBridgeDxe: IntersectMemoryDescriptor: desc [90000000, 90200000) type 1 cap 8700000000000001 conflicts with aperture [80000000, FD00000001) cap 1

ASSERT_EFI_ERROR (Status = Invalid Parameter)
ASSERT [PciHostBridgeDxe] /data/qiandongyan/test_1/test_2/uefi-loongarch-a2101/LsRefCodePkg/Bus/Pci/PciHostBridgeDxe/PciHostBridge.c(520): !EFI_ERROR (Status)
```

Can provide detailed hardware information if necessary.

kilaterlee, 2022-06-17:
This bug is known, we will fix it in next version @MarsDoge, **do not turn** this option on until the next release.

xen0n, 2022-06-17:
Thanks for the quick reply! I'll try removing cards first, to see if temporarily unplugging all cards would help (for avoiding having to dig out the flasher from some burial site).

kilaterlee, 2022-06-17:
> Thanks for the quick reply! I'll try removing cards first, to see if temporarily unplugging all cards would help (for avoiding having to dig out the flasher from some burial site).

In my opinion, this option is not must necessary because not all of PCIe device require 64-bit memory address space, 32bit will still work fine, so I suggest you just turn this option off until our fix it.

Chao,
Thanks.

xry111, 2022-06-17:
> > Thanks for the quick reply! I'll try removing cards first, to see if temporarily unplugging all cards would help (for avoiding having to dig out the flasher from some burial site).
> 
> In my opinion, this option is not must necessary because not all of PCIe device require 64-bit memory address space, 32bit will still work fine, so I suggest you just turn this option off until we fix it.
> 
> Chao, Thanks.

I guess Xuerui means he can't get into the configuration interface and turn off the option.  So he is trying to remove his PCIE devices.  If the removal still doesn't work he'll have to clear CMOS.

xen0n, 2022-06-17:
> In my opinion, this option is not must necessary because not all of PCIe device require 64-bit memory address space, 32bit will still work fine, so I suggest you just turn this option off until we [...]

Higher-end discrete GPUs typically benefit from such feature, because drivers would be able to directly address all of device memory without having to go through the 256M aperture. I plan to plug one of my GPUs in the future so it's good to verify functionality early.

xen0n, 2022-06-17:
> I guess Xuerui means he can't get into the configuration interface and turn off the option. So he is trying to remove his PCIE devices. If the removal still doesn't work he'll have to clear CMOS.

I think the configuration is written directly into the SPI flash, so no way around the flasher if removing cards doesn't work ;-)

kilaterlee, 2022-06-17:
> Higher-end discrete GPUs typically benefit from such feature, because drivers would be able to directly address all of device memory without having to go through the 256M aperture. I plan to plug [...]

Yes, usually the higher-end GPUs and AI controller needs require 64-bit memory, if you happen to use this kind device, it now appears to have to switch to a 32-bit require device in order to use it. :(

kilaterlee, 2022-06-17:
> I think the configuration is written directly into the SPI flash, so no way around the flasher if removing cards doesn't work ;-)

Yes, the CMOS resource are in short to supply, we will recommend the chips RD add more CMOS to use, and the key variables still stored in the flash now.

xen0n, 2022-06-17:
> Yes, usually the higher-end GPUs and AI controller needs require 64-bit memory, if you happen to use this kind device, it now appears to have to switch to a 32-bit require device in order to use it. [...]

They can work with Above 4G Decoding disabled, albeit with degraded performance, all drivers can access device memory via aperture because this resizable BAR feature is far from widespread.

I mean, it's beneficial for early adopters with adequate technical skills to help test this configuration, so we can even try to enable this by default for performance and maybe commercial advantage. After all, LoongArch is such a new platform, so we're uniquely able to raise our baseline very high (almost guaranteed Linux >= 5.19) and not fear breaking "older" systems.

xen0n, 2022-06-17:
Update: removing hardware didn't help, a flasher is definitely needed.

xry111, 2022-06-17:
> Update: removing hardware didn't help, a flasher is definitely needed.

So I'll purchase one before testing the new firmware :).

specialpointcentral, 2022-06-17:
> They can work with Above 4G Decoding disabled, albeit with degraded performance, all drivers can access device memory via aperture because this resizable BAR feature is far from widespread.
>
> I mean, [...]

So, what can I do if I want to disable `Above 4G Decoding`. I want to try it tonight, haha.

xen0n, 2022-06-17:
> So, what can I do if I want to disable `Above 4G Decoding`. I want to try it tonight, haha.

It's disabled by default. I only turned it on habitually after seeing it in the setup menu -- all my x86 systems have it enabled and I know Linux can handle it. So you should be fine if you don't randomly touch things in setup ;-)

kilaterlee, 2022-06-17:
> It's disabled by default. I only turned it on habitually after seeing it in the setup menu -- all my x86 systems have it enabled and I know Linux can handle it. So you should be fine if you don't [...]

Yes, don't touch this option at will, it will be fine :)

xen0n, 2022-06-18:
Notes for future visitors wanting to unbrick your A2101:

The SPI flash chip that comes with the A2101 is **Zbit ZB25VQ64ASIG** which supports SFDP, hence supported by the generic SPI flash code in `flashrom`, but you need to compile **the git version of flashrom** because it's conforming to SFDP v1.6 and parameter table is 64-byte-long, leading it to be rejected by older flashrom versions. I used a random CH341A flasher bought on Taobao and everything just worked.

Also you need to pad the `.fd` file to 8388608 bytes before flashing, I used 2 lines of python for that.

Good luck unbricking!

xen0n, 2022-08-04:
@kilaterlee Hi, may I ask when will the bug get fixed and a new beta pushed? Just a rough time frame would be enough I think.

xen0n, 2022-08-04:
> We are still focusing on the development of the new platform, I think it will be down soon and then we will fix thus bugs. Please wait for one week or two, thank you!

Thanks very much for the immediate reply, and your team's continued effort. Please just go ahead in your own pace.

(Apart from the status update people are expecting, I was pinging this thread mainly for confirming whether we can get an update before the upcoming LPC2022 event, so I would probably have something new kernel-side to experiment on.)

kilaterlee, 2022-08-04:
> Thanks very much for the immediate reply, and your team's continued effort. Please just go ahead in your own pace.
>
> (Apart from the status update people are expecting, I was pinging this thread [...]

Okay, we will try to fix them asap. :)

specialpointcentral, 2022-10-11:
Thats it. I got it. Firmware has been updated successfually. Thanks a lot.

MarsDoge, 2022-10-11:
> BTW could we get an ETA for the new firmware revision? Very eager to try if we can finally enable Above 4G Decoding, or if #2 is solved. ( ͡° ͜ʖ ͡°)

We fix it as soon as this weekend, thanks for the attention ! `v`

MarsDoge, 2022-10-11:
> BTW could we get an ETA for the new firmware revision? Very eager to try if we can finally enable Above 4G Decoding, or if #2 is solved. ( ͡° ͜ʖ ͡°)

We fix it as soon as this weekend, thanks for the attention !

xen0n, 2022-10-21:
Confirmed fixed by #11. Thanks!

closed 2022-10-21 by xen0n