loongson/Firmware #2: A2101 - cannot load grub2: TLB refill exception

xen0n, 2022-06-19:
```
FSOpen: Open '\EFI\arch\grubloongarch64.efi' Success
[Bds] Expand HD(1,GPT,DB9001E0-658D-F449-9CE4-80D374745795,0x800,0x80000)/\EFI\arch\grubloongarch64.efi -> PciRoot(0x0)/Pci(0x13,0x0)/Pci(0x0,0x0)/NVMe(0x1,42-00-1F-81-00-B7-28-A4)/HD(1,GPT,D
B9001E0-658D-F449-9CE4-80D374745795,0x800,0x80000)/\EFI\arch\grubloongarch64.efi
PROGRESS CODE: V03058000 I0
InstallProtocolInterface: 5B1B31A1-9562-11D2-8E3F-00A0C969723B FAD41040
Loading driver at 0x000F9E66000 EntryPoint=0x000F9E67000
InstallProtocolInterface: BC62157E-3E33-4FEC-9920-2D3B36D750DF FAD41E98
ProtectUefiImageCommon - 0xFAD41040
  - 0x00000000F9E66000 - 0x000000000003A000
PROGRESS CODE: V03058001 I0
TLB refill EPC  0xF9E6CA08
TLB refill BadVA  0x81C000030
```

Reproducible with both the grub in [LoongArch Arch Linux 2022.06.16.1 ISO](https://mirrors.wsyu.edu.cn/loongarch/2022.03/iso/2022.06/) and [my Gentoo grub package](https://github.com/xen0n/loongson-overlay/blob/master/sys-boot/grub/grub-2.11_pre20220617.ebuild).

@yetist says the LoongArch Arch Linux's grub should be working, so I guess this might be a firmware problem. Strange indeed...

yetist, 2022-06-20:
Is there any new message?

xen0n, 2022-06-20:
> Is there any new message?

The boot just hangs after the last 2 lines. They are apparently some kind of panic message.

xry111, 2022-06-20:
It's hard to tell if this is a firmware bug or a grub bug.

> Loading driver at 0x000F9E66000 EntryPoint=0x000F9E67000
> TLB refill EPC  0xF9E6CA08

If I interpret these two lines correctly you can take a look at the instruction in grubloongarch64.efi at 0x6a08 to understand the issue better.

xen0n, 2022-06-20:
> It's hard to tell if this is a firmware bug or a grub bug.
> 
> > Loading driver at 0x000F9E66000 EntryPoint=0x000F9E67000
> > TLB refill EPC  0xF9E6CA08
> 
> If I interpret these two lines correctly you can take a look at the instruction in grubloongarch64.efi at 0x6a08 to understand the issue better.

This grub is reported to be working on @yetist's board (CRB, not A2101) so I highly suspect A2101's firmware. I currently don't have enough mental energy to dive into this though.

kilaterlee, 2022-06-21:
> This grub is reported to be working on @yetist's board (CRB, not A2101) so I highly suspect A2101's firmware. I currently don't have enough mental energy to dive into this though.

@yetist has been tested Archlinux install an boot with Grub2 on A2101, and it work fine, but there is a bug about EfiFB when using the AMD graphics card, we will fix this in a few days.

xry111, 2022-06-30:
Interesting... I didn't see the issue using AMDGPU to output display.

xen0n, 2022-08-28:
I tested again with [updated grub sources](https://github.com/loongarch64/grub/commit/269385248955f2c6facc04a8ad1e638eff2b10c8), still the same error.

Any update on this?

xry111, 2022-08-28:
Interestingly, I can't reproduce this on my A2101...  even with AMDGPU installed.

Does A2101 has different sub models or it's only triggered by specific grub.cfg configuration?

xen0n, 2022-08-28:
> Does A2101 has different sub models (I can see some pads where no components are soldered on my board), or it's only triggered by specific grub.cfg configuration?

I don't know. My `grub.cfg` is just the auto-generated one from `grub-mkconfig`.

My board though is the very first batch of 3A5000 boards generally available, not sure if that'd make any difference. Waiting for the next beta firmware for fixed resizable BAR and other niceties anyway, so even if this one persists on my board not all is lost,

xry111, 2022-09-01:
> > Does A2101 has different sub models (I can see some pads where no components are soldered on my board), or it's only triggered by specific grub.cfg configuration?
> 
> I don't know. My `grub.cfg` is just the auto-generated one from `grub-mkconfig`.

Well, my `grub.cfg` is simply

```
set default=0
set timeout=5

insmod ext2
set root=(hd0,2)

menuentry "GNU/Linux, Linux 6.0.0-rc2+-lfs-cross-r11.1-241-systemd" {
        linux   /vmlinuz-6.0.0-rc2+-lfs-cross-r11.1-241-systemd root=/dev/sda3 ro console=ttyS0,115200
}
```

I guess I'll see the issue as well if I add `insmod all_video`.

xen0n, 2022-09-01:
> Well, my `grub.cfg` is simply
>
> ```
> set default=0
> set timeout=5
>
> insmod ext2
> set root=(hd0,2)
>
> menuentry "GNU/Linux, Linux 6.0.0-rc2+-lfs-cross-r11.1-241-systemd" {
>         linux [...]

Unfortunately it's not my case. I already have `GRUB_TERMINAL=console` in my `/etc/default/grub` and I don't even get to see the boot menu, it just instantly dies. I think I still have to wait for the next firmware revision.

xen0n, 2022-10-27:
Confirmed beta9 firmware is working with https://github.com/loongarch64/grub/commit/d18e9751ab11d5e0022d261ebbe8a4bae96c546e, thanks!

closed 2022-10-27 by xen0n