loongson/Firmware #6: A2101 - Crash running "ls" in grub shell after removing a USB stick

xry111, 2022-07-02:
When I booting the system with a USB stick plugged in, the boot failed because (hd0) changed from the HDD to the USB stick.  It's not an issue: such behavior is very common among all motherboards.  But when I removed the USB stick and tried to verify with "ls" in grub shell:

```
UsbRemoveDevice: device 1 removed
UsbEnumeratePort: device disconnected event on port 1

grub> ls
(proc) (hd0) 
SystemContext.SystemContextLoongArch address 0xFDE0BEB4
CsrCrmd   0xB0
CsrPrmd   0x4
CsrEctl  0x800
CsrEstat   0x480000
CsrEpc    0xF9E27B4C
CsrBadv    0xAFAFAFAFAFAFAFBF
CsrBadi 0x2400132B
Shut down slave cores done!
```

xen0n, 2022-07-12:
cc @yetist for grub port insights

kilaterlee, 2022-07-12:
> cc @yetist for grub port insights

Indeed, we found the real evidence in GRUB code that it not update or remove the descriptor timely manner.

yetist, 2022-07-13:
Is this a grub bug? Is it related to LoongArch?

xry111, 2022-07-13:
> Is this a grub bug? Is it related to LoongArch?

With GRUB 2.06 for x86_64 EFI `(hd0)` is also not removed in `ls` output, but it does not crash.

To me any application (not only GRUB) should print an error message when some unexpected error is hit (even if it's not recoverable), instead of just crash.

kilaterlee, 2022-07-13:
> With GRUB 2.06 for x86_64 EFI `(hd0)` is also not removed in `ls` output, but it does not crash.
>
> To me any application (not only GRUB) should print an error message when some unexpected error is [...]

Yes, it is not related to LoongArch. Please pay attention for this address: ```0xAFAFAFAFAFAFAFAF```, EDK2 will write this flag when the resouce be released, GRUB uses the release motherd of EDK2. GRUB code is not friendly for such operations, like quickly unplugging USB device and then accessing them. I guess it is not crash in X86 platform, might be they can handle addresses that like 0xAFAFAFxxxxx, but in LoongArch, the address is not accessible, it will be trigger TLB exceptions. Please @zwaizwai add more details. Thanks!

zwaizwai, 2022-07-13:
"grub> set debug=all"
"grub> ls"
You can view a more detailed debug info after typing these cmd.