loongson/Firmware #7: a2101: one PCIE slot does not work

xry111, 2022-07-02:
The PCIE x4 slot most close to CPU does not work.  Another PCIE x4 slot and the PCIE x16 slot work fine.

I've checked to BIOS setting, all PCIE ports are "enabled".

To make things worse, a two-slot GPU plugged in the PCIE x16 slot would physically block the only working PCIE x4 slot so I have no PCIE x4 slot to use now :(.

xen0n, 2022-10-11:
With hindsight this might actually be a hardware issue... not sure if this will get fixed in the next revision or confirmed unfixable.

xry111, 2022-10-11:
> With hindsight this might actually be a hardware issue... not sure if this will get fixed in the next revision or confirmed unfixable.

I'm not really sure too.  Now I've plugged an Ethernet cable in and removed the Wi-Fi card to use the GPU.

(A note here: the layout of PCIE slots on my board is not completely same as the A2101 photo provided in this repo.  Maybe it's the reason causing this issue.)