loongson/Firmware #8: Wonder about my motherboard type

jkxing, 2022-07-12:
My motherboard looks like ML5A, but board information shown in the system looks more like L5BMB01-CRB. I don't know which firmware suits my board.
[image]
[image]

kilaterlee, 2022-07-12:
You can use the ML5A firmware, the SMBIOS information is wrong, I guess this BIOS is not from original factory. It is recommended to store current BIOS binary before updating.

jkxing, 2022-07-17:
ML5A firmware works fine. Thanks

closed 2022-07-17 by jkxing