loongson/Firmware #9: ML5A: VGA and HDMI on board not working

yangxq, 2022-09-15:
VGA and HDMI on board have no signal. And I don't have an AMD card.
With the help of debug port, the Arch Linux can be installed and boot normally.

Are there any plans for a new release?

kilaterlee, 2022-09-15:
Yes, maybe this month or next month.

Thanks,
Chao
--------

MarsDoge, 2022-10-22:
> VGA and HDMI on board have no signal. And I don't have an AMD card. With the help of debug port, the Arch Linux can be installed and boot normally.
> 
> Are there any plans for a new release?

There are two models of ML5A, one must use a discrete graphics card and one can use a built-in graphics card, but the firmware required for both models is different.

loong64, 2022-10-22:
**ML5A—MB 无集显版  刷beta8的固件  SATA3接口掉硬盘   SATA2正常**
[image]

MarsDoge, 2022-10-22:
我今天改了一版ML5A-无集显版的prebeta9最新固件, 但却没机器来验证. -.-

loong64, 2022-10-22:
> 我今天改了一版ML5A-无集显版的prebeta9最新固件， 但却没机器来验证.-.-

你好，发一份我试试。 
710187964@qq.com

loong64, 2022-10-22:
> 我今天改了一版ML5A-无集显版的prebeta9最新固件, 但却没机器来验证. -.-
目前测试的问题：
1  开机主板蜂鸣器一直响不停止（可以开机）
2  sata3 重启还是有掉盘现象   （没规律）
3  主板插好U盘开机 选择启动维护管理 >文件引导  拔掉 U盘后    返回BIOS界面黑屏  有时直接死机
4  BIOS主板信息里不显示SATA设备

xry111, 2022-10-22:
> 3 主板插好U盘开机 选择启动维护管理 >文件引导 拔掉 U盘后 返回BIOS界面黑屏 有时直接死机

这个问题应该挺 common 的，A2101 好像也有，只能建议先不要在开着设置界面的时候拔 U 盘。

MarsDoge, 2022-10-22:
> > 1  开机主板蜂鸣器一直响不停止（可以开机）
       可以先尝试在UEFI shell下执行两条命令先尝试关闭： 1. mm 0xe00100e0800 -w 1 0     2.mm 0xe00100e0900 -w 1 0
> > 2  sata3 重启还是有掉盘现象   （没规律）
       这可能是ML5A板卡自身的问题. 
> > 3  主板插好U盘开机 选择启动维护管理 >文件引导  拔掉 U盘后    返回BIOS界面黑屏  有时直接死机
       这个问题如果还出现有log的话帮忙提供下，如果是grub引导阶段拔U盘这个问题我们深入debug一下.
> > 4  BIOS主板信息里不显示SATA设备
       该问题已知，主板信息Setup界面下信息没有与该板卡做适配，所以有可能会出现该问题，但不会影响功能的，ML5A开发固件只能先这样了.

loong64, 2022-10-22:
> > > 1  开机主板蜂鸣器一直响不停止（可以开机）
> > > 可以先尝试在UEFI shell下执行两条命令先尝试关闭： 1. mm 0xe00100e0800 -w 1 0     2.mm 0xe00100e0900 -w 1 0
输入代码后 蜂鸣器还是一直响  重启也一样 
> > > 2  sata3 重启还是有掉盘现象   （没规律）
> > > 这可能是ML5A板卡自身的问题.
好的
> > > 3  主板插好U盘开机 选择启动维护管理 >文件引导  拔掉 U盘后    返回BIOS界面黑屏  有时直接死机
> > > 这个问题如果还出现有log的话帮忙提供下，如果是grub引导阶段拔U盘这个问题我们深入debug一下.
好的
> > > 4  BIOS主板信息里不显示SATA设备
> > > 该问题已知，主板信息Setup界面下信息没有与该板卡做适配，所以有可能会出现该问题，但不会影响功能的，ML5A开发固件只能先这样了.
好的

MarsDoge, 2022-10-22:
> > > > 1  开机主板蜂鸣器一直响不停止（可以开机）
> > > > 可以先尝试在UEFI shell下执行两条命令先尝试关闭： 1. mm 0xe00100e0800 -w 1 0     2.mm 0xe00100e0900 -w 1 0
> > > > 输入代码后 蜂鸣器还是一直响  重启也一样

 1） mm 0xe00100e0800 -w 1 0     2）mm 0xe00100e0900 -w 1 1    
       *步骤2最后的 0 改成 1 再尝试一下，我这儿没板卡和原理图所以只能请再验证一下，Tks*

loong64, 2022-10-22:
> > > > > 1  开机主板蜂鸣器一直响不停止（可以开机）
> > > > > 可以先尝试在UEFI shell下执行两条命令先尝试关闭： 1. mm 0xe00100e0800 -w 1 0     2.mm 0xe00100e0900 -w 1 0
> > > > > 输入代码后 蜂鸣器还是一直响  重启也一样
> 
> 1） mm 0xe00100e0800 -w 1 0 2）mm 0xe00100e0900 -w 1 1 _步骤2最后的 0 改成 1 再尝试一下，我这儿没板卡和原理图所以只能请再验证一下，Tks_

没事  还是不行  
[image]

closed 2022-10-24 by kilaterlee