loongson/Firmware #16: 3a5000 7a2000 evb固件请求

FanFansfan, 2022-11-06:
smbios信息

`# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Table at 0x90000000FE360000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: Loongson
	Version: Loongson-UDK2018-V2.0.05269-beta8
	Release Date: Unknown
	ROM Size: 4096 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		Serial services are supported (int 14h)
		USB legacy is supported
		UEFI is supported
	BIOS Revision: 2.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Loongson
	Product Name: Loongson-3A5000-HV-7A2000-1w-V0.1-EVB
	Version: Not Specified
	Serial Number: Not Specified
	UUID: Not Present
	Wake-up Type: Power Switch
	SKU Number: Not Specified
	Family: Not Specified

Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
	Manufacturer: Loongson
	Product Name: Loongson-LS3A5000-7A2000-1w-EVB-V1.21
	Version: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Specified
	Chassis Handle: 0x0000
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 24 bytes
Chassis Information
	Manufacturer: Loongson
	Type: Desktop
	Lock: Not Present
	Version: Not Specified
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: Unknown
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: Unspecified
	Contained Elements: 0
	SKU Number: Not Specified

Handle 0x0004, DMI type 4, 48 bytes
Processor Information
	Socket Designation: CPU1
	Type: Central Processor
	Family: Unknown
	Manufacturer: Loongson
	ID: 33 41 35 30 30 30 2D 48
	Version: Loongson-3A5000-HV
	Voltage: 1.2 V
	External Clock: 25 MHz
	Max Speed: 2500 MHz
	Current Speed: 2500 MHz
	Status: Populated, Enabled
	Upgrade: None
	L1 Cache Handle: Not Provided
	L2 Cache Handle: Not Provided
	L3 Cache Handle: Not Provided
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: Not Specified
	Core Count: 4
	Core Enabled: 4
	Thread Count: 4
	Characteristics:
		64-bit capable
		Multi-Core
		Hardware Thread

Handle 0x0005, DMI type 7, 27 bytes
Cache Information
	Socket Designation: Not Specified
	Configuration: Enabled, Not Socketed, Level 1
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 64 kB
	Maximum Size: 64 kB
	Supported SRAM Types:
		Burst
		Pipeline Burst
		Synchronous
	Installed SRAM Type: Burst Pipeline Burst Synchronous
	Speed: Unknown
	Error Correction Type: Single-bit ECC
	System Type: Data
	Associativity: 4-way Set-associative

Handle 0x0006, DMI type 7, 27 bytes
Cache Information
	Socket Designation: Not Specified
	Configuration: Enabled, Not Socketed, Level 2
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 256 kB
	Maximum Size: 256 kB
	Supported SRAM Types:
		Burst
		Pipeline Burst
		Synchronous
	Installed SRAM Type: Burst Pipeline Burst Synchronous
	Speed: Unknown
	Error Correction Type: Single-bit ECC
	System Type: Data
	Associativity: 16-way Set-associative

Handle 0x0007, DMI type 7, 27 bytes
Cache Information
	Socket Designation: Not Specified
	Configuration: Enabled, Not Socketed, Level 3
	Operational Mode: Write Back
	Location: Internal
	Installed Size: 16384 kB
	Maximum Size: 16384 kB
	Supported SRAM Types:
		Burst
		Pipeline Burst
		Synchronous
	Installed SRAM Type: Burst Pipeline Burst Synchronous
	Speed: Unknown
	Error Correction Type: Single-bit ECC
	System Type: Data
	Associativity: 16-way Set-associative

Handle 0x0008, DMI type 9, 17 bytes
System Slot Information
	Designation: PCIE X4 SLOT
	Type: x4 PCI Express 2 x4
	Current Usage: Available
	Length: Short
	ID: 0
	Characteristics:
		3.3 V is provided

Handle 0x0009, DMI type 9, 17 bytes
System Slot Information
	Designation: PCIE X8 SLOT 1
	Type: x8 PCI Express 2 x8
	Current Usage: Available
	Length: Short
	ID: 1
	Characteristics:
		3.3 V is provided

Handle 0x000A, DMI type 9, 17 bytes
System Slot Information
	Designation: PCIE X8 SLOT 2
	Type: x8 PCI Express 2 x8
	Current Usage: Available
	Length: Short
	ID: 2
	Characteristics:
		3.3 V is provided

Handle 0x000B, DMI type 9, 17 bytes
System Slot Information
	Designation: PCIE X8 SLOT 3
	Type: x8 PCI Express 2 x8
	Current Usage: In Use
	Length: Short
	ID: 3
	Characteristics:
		3.3 V is provided
	Bus Address: 0000:05:00.0

Handle 0x000C, DMI type 11, 5 bytes
OEM Strings
	String 1: Standard-Loongson

Handle 0x000D, DMI type 16, 23 bytes
Physical Memory Array
	Location: System Board Or Motherboard
	Use: System Memory
	Error Correction Type: None
	Maximum Capacity: 16 GB
	Error Information Handle: Not Provided
	Number Of Devices: 2

Handle 0x000E, DMI type 19, 31 bytes
Memory Array Mapped Address
	Starting Address: 0x0000000000000000k
	Ending Address: 0x0000000800000000k
	Range Size: 32 GB
	Physical Array Handle: 0x0000
	Partition Width: 0

Handle 0x000F, DMI type 28, 22 bytes
Temperature Probe
	Description: Cpu Temperature
	Location: Processor
	Status: OK
	Maximum Value: 128.0 deg C
	Minimum Value: Unknown
	Resolution: Unknown
	Tolerance: Unknown
	Accuracy: Unknown
	OEM-specific Information: 0x00000000
	Nominal Value: 0.0 deg C

Handle 0x0010, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0011, DMI type 39, 22 bytes
System Power Supply
	Power Unit Group: 1
	Location: To Be Filled By O.E.M
	Name: To Be Filled By O.E.M
	Manufacturer: To Be Filled By O.E.M
	Serial Number: To Be Filled By O.E.M
	Asset Tag: To Be Filled By O.E.M
	Model Part Number: To Be Filled By O.E.M
	Revision: To Be Filled By O.E.M
	Max Power Capacity: Unknown
	Status: Present, Unknown
	Type: Unknown
	Input Voltage Range Switching: Unknown
	Plugged: No
	Hot Replaceable: Yes

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
`

xry111, 2022-11-06:
顺便问一下，现在 7A2000 有内核驱动吗？

xen0n, 2022-11-06:
> 顺便问一下，现在 7A2000 有内核驱动吗？

Same as 7A1000's

MarsDoge, 2022-11-10:
@FanFansfan 方便的话看一下主板上的丝印是哪个版本的, 换句话说是v1.4以上还是v1.4以下的版本？

FanFansfan, 2022-11-11:
> @FanFansfan 方便的话看一下主板上的丝印是哪个版本的, 换句话说是v1.4以上还是v1.4以下的版本？
v1.5的，谢谢。

KatyushaScarlet, 2022-11-11:
> @FanFansfan 方便的话看一下主板上的丝印是哪个版本的, 换句话说是v1.4以上还是v1.4以下的版本？

請問1.4及以下的版本，與1.4以上的版本，有哪些區別？

FanFansfan, 2022-11-13:
测试能启动loongnix和archlinux了。

liangyongxiang, 2022-11-19:
我这里测试不能工作，原来是有loognix系统的，现象是
- hdmi和uart都没有输出，其中uart我测试过板子的TX/RX接ttl的RX。
- 不接内存条会有警告
- 风扇转速很慢

MarsDoge, 2022-11-20:
> 我这里测试不能工作，原来是有loognix系统的，现象是
> 
> * hdmi和uart都没有输出，其中uart我测试过板子的TX/RX接ttl的RX。
> * 不接内存条会有警告
> * 风扇转速很慢
> 
> 等了几分钟之后确实可以了。

是的，7A2000平台第一次启动，确实需要一点时间. 
REF: https://github.com/loongson/Firmware/pull/18#issuecomment-1318912826

FanFansfan, 2022-11-20:
> 我这里测试不能工作，原来是有loognix系统的，现象是
> 
> * hdmi和uart都没有输出，其中uart我测试过板子的TX/RX接ttl的RX。
> * 不接内存条会有警告
> * 风扇转速很慢
> 
> 等了几分钟之后确实可以了。

uart要rs232转ttl，再接ttl

donmor, 2022-11-25:
话说天创者的l5a2是不是就是这个板子🤔

MarsDoge, 2022-11-25:
> 话说天创者的l5a2是不是就是这个板子🤔

是的

closed 2022-12-05 by kilaterlee