loongson/Firmware #23: Booting new world open source Linux ditro on 3A5000-7A2000-EVB

mikhailnov, 2023-01-29:
Hello chinese friends!

I have motherboard 3A5000-7A2000-EVB v1.5 with UEFI (BIOS) firmware Loongson-UDK2018-V2.0.05269-beta8 which seems to be much older than DK2018_3A5000-7A2000_Desktop_EVB_V4.0.05383-beta10_dbg.fd in this repository.
https://linux-hardware.org/?probe=2fea9476f5

I have successfully booted and installed Loongnix 20. It is an "old world" Linux distribution built with non-upstream toolchain. This article https://blog.xen0n.name/en/posts/tinkering/loongarch-faq says about new and old worlds (why exactly they are incompatible, I have not understood yet). Now I want to boot an open source new world Linux distro built with upstream toolchain (gcc, binutils). But I cannot boot any of them.

First, I tried to boot LiveCDs , e.g. https://github.com/fedora-remix-loongarch/releases-info , in virt-manager (qemu) which is preinstalled in Loongnix and got "cannot allocate memory for memory map"

I tried to add initrd+vmlinuz from them into Loognix's Grub bootloader. It froze when trying to load them.

Then I tried to replace `/boot/efi/EFI/*/grub*.efi` with Grub EFI binary from a new world LiveCD and got the same error, now on the real machine:
[image]

The only thing that comes to mind is that this UEFI (BIOS) version does not support new ISA/ABI (how to name it correctly?).
There is no "Security" page in the UEFI, so updating via a EFI shell will probably be the only option. But I am afraid to brick the motherboard.

Could you please comment if updating firmware is the right solution to run new world LoongArch binaries?
Thanks)

mikhailnov, 2023-01-29:
After studying https://lpc.events/event/16/contributions/1178/attachments/1000/1948/lpc2022-loongarch-20220913.pdf I am pretty sure that UEFI firmware will be updated, but will the new formware from this repository be able to boot both old and new world Linux distributions (e.g. both Loongnix 20 and Fedora)?

MarsDoge, 2023-01-30:
> After studying https://lpc.events/event/16/contributions/1178/attachments/1000/1948/lpc2022-loongarch-20220913.pdf I am pretty sure that UEFI firmware must be updated, but will the new formware from this repository be able to boot both old and new world Linux distributions (e.g. both Loongnix 20 and Fedora)?

As a description, support for booting an open source "new world" Linux distro does require firmware updates.
The new firmware can be downgraded to support the "old world", but only if this option is turned on under the BIOS interface Advance/Legacy.

xen0n, 2023-01-30:
> The new firmware can be *downgraded* to support the "old world"

Probably you mean just "it can be made to support the old world", a downgrade is not required given you have the boot mode toggle. (I haven't tried out the latest A2101 FW revision myself yet.)

And yes @mikhailnov you should definitely upgrade to the latest revision if it indeed has the boot mode switch, so you can indeed dual-boot both worlds. Newer Loongnix is confirmed to boot on both OW and NW firmware.

MarsDoge, 2023-03-31:
If you have tried successfully, please close this issue, thanks for your attention! @mikhailnov

closed 2023-04-11 by MarsDoge

matoro, 2023-07-30:
By the way, I know this is a closed issue, but I would just like to add a note since I ran into this same problem.  The factory firmware can output on either the HDMI port or the VGA port, but the new firmware can ONLY output on the VGA port.  I thought I had screwed up the firmware upgrade, but the problem is that running the new firmware requires a VGA monitor - it will not work on the HDMI port.

mikhailnov, 2024-08-01:
Thanks, upgrading to https://github.com/loongson/Firmware/blob/main/5000Series/PC/3A5000-7A2000-EVB/UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05569-stable202308_rel.fd has helped, both old world Loongnix and new world ALT Linux boot correctly without any switching in UEFI settings.
@matoro HDMI works ok