loongson/Firmware #26: 用这个固件在qemu上启动自己的内核，进入了uefi shell，如何进一步引导呢

GooTal, 2023-02-26:
我编了一个nuttx内核，龙芯架构。
qemu命令：
```
./qemu-system-loongarch64 -m 4G -smp 1 --cpu la464 --machine virt -bios ./bios/edk2-loongarch64-code.fd -kernel ./nuttx -vga none -nographic -gdb tcp::1122 -S
```
看起来只能进入uefi shell，这个内核没有文件系统，直接用-kernel指定了内核。
启动以后，进入了uefi shell，请问接下来如何引导内核？
[image]

xry111, 2023-02-26:
目力看了一下您的提交 https://github.com/LA-NuttX/la-nuttx/commit/f0c3507af74b32d767a3443488e1ea1f645a6604 貌似没有 EFI stub (当然也可能我瞎了)，https://github.com/apache/nuttx/blob/master/arch/arm64/src/common/arm64_head.S 有 ARM64 的可以参考一下。

GooTal, 2023-02-26:
感谢回复。
抱歉，刚提交了一个[版本](https://github.com/apache/nuttx/commit/4440275a34a13b1bcacf8ddee1d41e2c0a65bd4d)。
这个版本编译出的内核，可以跑在qemu-system-looongarch64 v6.2版本上， v6.2具体运行环境是https://github.com/foxsen/qemu-loongarch-runenv，使用的仓库里的bios。

最近调timer 和 uart中断，发现6.2版本的qemu里，timer清中断把uart中断清掉了。
看到最新版本的qemu好像修复了，于是，尝试拉取编译了官方qemu-system-loongarch64 V7.2.0版本的qemu：
[image]
然后拉取编译了bios，bios编译方法按照：https://github.com/foxsen/qemu-loongarch-runenv/blob/devel/docker/setup-bios.sh。

参考[该仓库](https://github.com/foxsen/qemu-loongarch-runenv/tree/devel/docker)，我使用了如下命令，尝试启动nuttx:
```
./qemu-system-loongarch64 -machine virt -m 4G -cpu la464-loongarch-cpu\
	-smp 1\
	-bios ./bios/QEMU_EFI.fd\
	-kernel ./nuttx\
	--drive id=disk0,file=./poky-minimal.ext4,if=none,format=raw -device virtio-blk-pci,drive=disk0\
	-append "root=/dev/vda rdinit=/sbin/init console=ttyS0,115200"\
	--nographic\
	-gdb tcp::1122 -S
```
启动后进入uefi shell：
[image]

BTW，我上传了这个loongarch版本[nuttx内核](https://github.com/LA-NuttX/runnable-binay/blob/main/nuttx)，它只用了映射窗口（nuttx的flat build模式没有启用页表），信息如下：
[image]

想请教，如何在UEFI shell进一步启动我的nuttx内核，它只有一个内核，没有额外的文件系统。

谢谢！

xen0n, 2023-02-26:
没看错的话，你的构建产物现在是个 ELF 文件？顾名思义，UEFI 固件它需要的是一个 EFI application。

具体可以参考下 [grub2](https://github.com/loongarch64/grub/tree/dev/patchwork/efi) 或者 Linux EFI stub 他们在龙架构都怎么做。

xen0n, 2023-02-26:
追一下 qemu 源码（`hw/loongarch/virt.c`，关键词 kernel），你那么调用之后直接进了 EFI shell 应该就是固件不认。他们现在貌似做了两个引导流程，有固件时候不处理 kernel image 直接通过 fw_cfg 机制交固件处理；没固件时候假定是 ELF 格式内核（结合龙芯自己商业上的安排，预期估计是 ELF 内核就是 DT boot，但显然他们没做区分，无论有无固件都初始化了 ACPI 和 FDT）。

GooTal, 2023-02-26:
hello，谢谢提醒 @xen0n ，我发现qemu7.2的hw/loongarch/virt.c中，loongarch_init调用了loongarch_direct_kernel_boot函数，此处是bios指定为空的情况，也就是说，我们根本不用指定bios辣，只要-kernel就行了。

这是我的运行命令：
```
./qemu-system-loongarch64 
    -machine virt 
    -m 4G 
    -cpu la464
    -smp 1 
    -kernel ./nuttx
    --nographic
    -gdb tcp::1122 -S
```

另外，发现了qemu7.2的hw/loongarch/virt.c文件的cpu_loongarch_virt_to_phys函数中，它只保留了低29位？之前是保留32位来着，此处不太理解改动原因。我暂时还是将此处改回32位。（我的nuttx内核是 flat build模式，不启用页表，链接脚本的地址指定了0x9000 0000 a000 0000)，也就被装入了0xa000 0000的物理地址，如果只保留低29位，可能会写乱内存其他地方）。
不知道改为32位是否会有其他问题。
Anyway，跑起来了：
[image]
暂时还有些小bug等待修复。谢谢大伙。