loongson/Firmware #27: 使用edk2固件在qemu启动出错

prcups, 2023-02-27:
使用环境为x86_64 archlinux，qemu版本7.2
无论是使用自己编译的edk2固件还是这个仓库里的版本都会报错：
仓库的版本：
[image]
自编译版本：
[image]
使用[这个版本](https://github.com/yangxiaojuan-loongson/qemu-binary)正常启动，但是map中没有fs0设备（虽然能正常检测到AHCI控制器）

driver1998, 2023-02-28:
最新的edk2加了pflash支持，对应qemu在这个commit
https://github.com/qemu/qemu/commit/288431a1fb9334d5d57ad7d5854d8475b23e7c42
7.2.0是没有的

prcups, 2023-02-28:
现在可以正常启动了，但是显示No Mapping Found
使用的启动参数如下：
```
qemu-system-loongarch64 -m 1G \
--cpu la464-loongarch-cpu \
--machine virt \
-usb \
-bios /usr/share/edk2-ovmf/loongarch64/OVMF_CODE.fd  \
-device usb-ehci,id=ehci \
-drive id=disk,file=a.img,format=raw,if=none \
-device ahci,id=ahci \
-device ide-hd,drive=disk,bus=ahci.0 \
-device usb-kbd \
-device ramfb
```
是因为AHCI这块还没做支持吗

driver1998, 2023-02-28:
要不试试virtio-blk或者virtio-scsi？loongarch qemu 的固件好像没编译AHCI驱动。

xen0n, 2023-02-28:
@prcups 可以试试把串口重定向出来，提供下日志，这样龙芯团队也方便查原因？

prcups, 2023-02-28:
串口只有我自己编译的版本有输出（大概是没关调试）：
https://yuno.krgm.moe:29443/s/bNywDso7xxBTHmX

prcups, 2023-02-28:
> 要不试试virtio-blk或者virtio-scsi？ ~loongarch qemu 的固件好像没编译AHCI驱动。~ 看漏了

意思是可以自己编译吗？
换协议需要重新写驱动，有些麻烦（

driver1998, 2023-02-28:
https://github.com/loongson/edk2-platforms/blob/88d44c563d9fd5c95be93e706f9420352ee4c053/Platform/Loongson/LoongArchQemuPkg/Loongson.dsc#L550

感觉应该是有的，我刚刚没看到

xen0n, 2023-02-28:
> > 要不试试virtio-blk或者virtio-scsi？ ~loongarch qemu 的固件好像没编译AHCI驱动。~ 看漏了
> 
> 意思是可以自己编译吗？ 

这个指的是自己编译啥？

> 换协议需要重新写驱动，有些麻烦（

virtio 专用于虚拟化场景的，应该适配起来更简单才对

prcups, 2023-02-28:
刚刚试了下virtio能用，我先这么用着吧

xen0n, 2023-02-28:
> 刚刚试了下virtio能用，我先这么用着吧

可以

不过 issue 建议留着先不关，先前使用姿势的问题还是要查的

prcups, 2023-02-28:
好（

MarsDoge, 2025-11-04:
关于在上游 edk2 虚拟机固件Ovmf提交支持 AHCI 的 SATA 设备：https://github.com/tianocore/edk2/pull/11309 已经合并到tag-stable2408基线;