loongson/Firmware #28: 联想的开天M540Z使用 UDK2018-LoongArch-CRB-pre-beta9.fd 无法开机

liangyongxiang, 2023-02-28:
硬件：https://item.jd.com/100025325285.html
BIOS信息：
[image]

dmidecode输出：
```
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Table at 0x90000000FD77C000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: ZD-TECH
	Version: W05KT1CA
	Release Date: 12/31/2021
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		Serial services are supported (int 14h)
		USB legacy is supported
		Function key-initiated network boot is supported
		UEFI is supported
	BIOS Revision: 4.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Loongson
	Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
	Version: 1.0
	Serial Number: TBD by OEM
	UUID: 00112233-4455-6677-8899-aabbccddeeff
	Wake-up Type: Power Switch
	SKU Number: Loongson_SKU
	Family: Type1Family

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Loongson
	Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
	Version: 1.0
	Serial Number: MX20BA001J
	Asset Tag: Type2 - Board Asset Tag
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Type2 - Board Chassis Location
	Chassis Handle: 0xFFFF
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Loongson
	Type: Rack Mount Chassis
	Lock: Not Present
	Version: Version To Be Filled By O.E.M
	Serial Number: Chassis Board Serial#To Be Filled By O.E.M
	Asset Tag: Chassis Board Asset Tag#To Be Filled By O.E.M
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
[... 341 lines cut ...]
	Designation: PCIE16X1
	Type: x16 Other
	Current Usage: In Use
	Length: Other
	Characteristics: Unknown
	Bus Address: 0000:05:00.0

Handle 0x001E, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: SATA Controller
	Type: SATA Controller
	Status: Enabled
	Type Instance: 1
	Bus Address: 0000:00:08.0

Handle 0x001F, DMI type 10, 6 bytes
On Board Device Information
	Type: SATA Controller
	Status: Enabled
	Description: SATA Controller

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table

```

MarsDoge, 2023-03-01:
联想开天M540Z的板型不是CRB类型板卡，Firmware暂时无该板卡型号支持.  @liangyongxiang

xen0n, 2023-03-01:
> 联想开天M540Z的板型不是CRB类型板卡，Firmware暂时无该板卡型号支持. @liangyongxiang

这是巨坑，显然联想的固件人违反了[《龙芯CPU统一系统架构规范》](https://github.com/loongson-community/docs/blob/master/firmware/%E9%BE%99%E8%8A%AFCPU%E7%BB%9F%E4%B8%80%E7%B3%BB%E7%BB%9F%E6%9E%B6%E6%9E%84%E8%A7%84%E8%8C%83-v4.0.pdf)。建议修订一下此处的 README 避免更多人掉坑。。。。不是所有人都会听从建议备份固件！

closed 2023-11-30 by MarsDoge