loongson/Firmware #29: beta10启动需要很久才会显示Logo界面

haowuge, 2023-03-04:
天创者主板，物理机刷了 bata10 后启动时需要等很久才会显示Logo界面，
是否有什么方法可以提高启动速度？谢谢
```
BIOS Information
        Vendor: Loongson
        Version: Loongson-UDK2018-V4.0.05383-beta10
        Release Date: 11/30/2022
        ROM Size: 4 MB
        Characteristics:
```

xen0n, 2023-03-04:
估计是因为开了很多串口调试信息的缘故，是不是可以考虑提供一些release版的固件？用户可以默认刷这版本，有问题再去刷debug版本协助调试。

MarsDoge, 2023-03-06:
> 估计是因为开了很多串口调试信息的缘故，是不是可以考虑提供一些release版的固件？用户可以默认刷这版本，有问题再去刷debug版本协助调试。

可以提供release版, 我猜测 @haowuge 的现象应该是第一次固件刷新后启动需要训练显存参数的缘故;

haowuge, 2023-03-07:
> 

重启开了好几次了，每次都慢
现在加了R7 240也慢

haowuge, 2023-03-10:
> > 估计是因为开了很多串口调试信息的缘故，是不是可以考虑提供一些release版的固件？用户可以默认刷这版本，有问题再去刷debug版本协助调试。
> 
> 可以提供release版, 我猜测 @haowuge 的现象应该是第一次固件刷新后启动需要训练显存参数的缘故;

> > 估计是因为开了很多串口调试信息的缘故，是不是可以考虑提供一些release版的固件？用户可以默认刷这版本，有问题再去刷debug版本协助调试。
> 
> 可以提供release版, 我猜测 @haowuge 的现象应该是第一次固件刷新后启动需要训练显存参数的缘故;

所以大概啥时候会有release版本？
顺便问个，主板自带的显卡有驱动了没？

closed 2023-03-14 by kilaterlee