loongson/Firmware #30: Firmware for LS2C50C2 board

asheplyakov, 2023-03-06:
```
Handle 0x0001, DMI type 0, 26 bytes
BIOS Information
        Vendor: Loongson
        Version: Loongson-UDK2018-V2.2.6-8W-pcba
        Release Date: 09/23/2022
        ROM Size: 4 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                Serial services are supported (int 14h)
                USB legacy is supported
                Function key-initiated network boot is supported
                UEFI is supported
        BIOS Revision: 2.2

Handle 0x0003, DMI type 2, 17 bytes               
Base Board Information                                                                         
        Manufacturer: LOONGSON                                                                 
        Product Name: Loongson-LS2C50C2                                                        
        Version: LS2C50C2                                                                      
        Serial Number: XXXXXXXXXXXXXXXXXX                                                      
        Asset Tag: To be filled by O.E.M.To be fill                                            
        Features:                                                                              
                Board is a hosting board                                                       
                Board is replaceable                                                           
        Location In Chassis: Not Specified                                                     
        Chassis Handle: 0x0000                                                                 
        Type: Motherboard                                                                      
        Contained Object Handles: 0 

Handle 0x0008, DMI type 4, 48 bytes
Processor Information
        Socket Designation: CPU1
        Type: Central Processor
        Family: Unknown
        Manufacturer: Loongson
        ID: 33 43 35 30 30 30 00 00
        Version: Loongson-3C5000
        Voltage: 1.2 V
        External Clock: 25 MHz
        Max Speed: 1600 MHz
        Current Speed: 2200 MHz
        Status: Populated, Enabled
        Upgrade: None
        L1 Cache Handle: 0x0005
        L2 Cache Handle: 0x0006
        L3 Cache Handle: 0x0007
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Core Count: 16
        Core Enabled: 16
        Thread Count: 16
        Characteristics:
                64-bit capable
                Multi-Core
                Hardware Thread

Handle 0x000C, DMI type 4, 48 bytes
Processor Information
        Socket Designation: CPU2
        Type: Central Processor
        Family: Unknown
        Manufacturer: Loongson
        ID: 33 43 35 30 30 30 00 00
        Version: Loongson-3C5000
        Voltage: 1.2 V
        External Clock: 25 MHz
        Max Speed: 1600 MHz
        Current Speed: 2200 MHz
        Status: Populated, Enabled
        Upgrade: None
        L1 Cache Handle: 0x0009
        L2 Cache Handle: 0x000A
        L3 Cache Handle: 0x000B
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Part Number: Not Specified
        Core Count: 16
        Core Enabled: 16
        Thread Count: 16
        Characteristics:
                64-bit capable
                Multi-Core
                Hardware Thread
```

MarsDoge, 2023-03-14:
fixes: https://github.com/loongson/Firmware/pull/31
@asheplyakov Are there any other questions? If not, please close it. Thank you !

asheplyakov, 2023-03-14:
The firmware appears to work fine with the **old world** kernel.
However the new world one (loongarch-next commit 90864195ebe99ac688ab79444fb787ab72996144) can boot only with `maxcpus=16` (otherwise it locks up when bringing up CPU 16).
Not sure if this is a kernel issue or the firmware one.

Perhaps I'll close this issue anyway

MarsDoge, 2023-03-14:
> The firmware appears to work fine with the **old world** kernel. However the new world one (loongarch-next commit 90864195ebe99ac688ab79444fb787ab72996144) can boot only with `maxcpus=16` (otherwise [...]

Please check whether the Advance/LegacyMode of setup is closed, if not, please close it, and then boot the new world.