loongson/Firmware #34: 新固件DMI信息内存部分缺少内存运行频率相关信息

haowuge, 2023-03-23:
天创者主板，新固件DMI信息 内存部分 缺少内存运行频率相关的详细信息（比如缺少运行频率），是否能增加这部分信息？谢谢

Speed: 3200 MT/s
Configured Memory Speed: 3200 MT/s

xen0n, 2023-03-23:
FYI:

A2101 prebeta10 DMI 表格种类

```
$ sudo dmidecode | grep -o 'DMI type [0-9]\+' | sort -n -k 3 | uniq
DMI type 0
DMI type 1
DMI type 2
DMI type 3
DMI type 4
DMI type 7
DMI type 9
DMI type 11
DMI type 16
DMI type 19
DMI type 28
DMI type 32
DMI type 39
DMI type 44
DMI type 127
```

旧世界昆仑固件：

```
# ZD-TECH KL.4.1H.LM.D.014.210521.R
$ cat ~/dmidecode.bpi.log | grep -o 'DMI type [0-9]\+' | sort -n -k 3 | uniq
DMI type 0
DMI type 1
DMI type 2
DMI type 3
DMI type 4
DMI type 7
DMI type 9
DMI type 11
DMI type 13
DMI type 16
DMI type 17
DMI type 19
DMI type 32
DMI type 127
```

比较二者可发现：prebeta10 固件缺了 type 13 (BIOS Language) 和 17 (Memory Device) 两种。本 issue 就是需要暴露 Memory Device 类型的表格。

haowuge, 2023-08-16:
更新到 UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05536-Prestable202308_dbg.fd ，可以正常显示内存频率了

```
Handle 0x000F, DMI type 17, 40 bytes
Memory Device
	Array Handle: 0x0000
	Error Information Handle: Not Provided
	Total Width: 64 bits
	Data Width: 64 bits
	Size: 8 GB
	Form Factor: DIMM
	Set: None
	Locator: A2
	Bank Locator: Bank
	Type: DDR4
	Type Detail: Registered (Buffered)
	Speed: 2666 MT/s
	Manufacturer: <BAD INDEX>
	Serial Number: <BAD INDEX>
	Asset Tag: <BAD INDEX>
	Part Number: <BAD INDEX>
	Rank: 1
	Configured Memory Speed: 2600 MT/s
	Minimum Voltage: 1.2 V
	Maximum Voltage: 1.2 V
	Configured Voltage: 1.2 V
```

closed 2023-11-04 by kilaterlee