loongson/Firmware #35: 增加Loongson-3A5000-7A2000-1w-V0.1-CRB的新世界固件请求

wszqkzqk, 2023-03-31:
系统信息：
```unix-conf
System Information
        Manufacturer: Loongson
        Product Name: Loongson-LS3A5000-7A2000-1w-V0.1-CRB
        Version: 1.0
        Serial Number: TBD by OEM
        UUID: 00112233-4455-6677-8899-aabbccddeeff
        Wake-up Type: Power Switch
        SKU Number: Loongson_SKU
        Family: Type1Family
```

MarsDoge, 2023-03-31:
@wszqkzqk 请确认您的CRB板卡是否与本次提交的板卡照片(Image/LSA5A2P2.jpeg)一致，如不一致，勿烧录，并上传照片反馈进行确认.
https://raw.githubusercontent.com/loongson/Firmware/c5f670dafc01357934c07fcaa73a31c733fd25fa/Image/LSA5A2P2.jpeg

wszqkzqk, 2023-03-31:
[image]

wszqkzqk, 2023-03-31:
请问可以使用w25q64fw吗

MarsDoge, 2023-04-01:
@wszqkzqk 您上传的2000主板非CRB标准板卡，还请您先与主板或整机提供厂商联系是否有相应固件提供更新(BIOS主线版本: Beta10或stable202302均可)，如您获取不到，请和我们反馈，我们再协调为您支持, 补充w25q64fw的flash型号是否支持还需和厂商确认(存在3.3v和1.8v区别)，我们目前没该板卡原理图,  感谢理解 !

AnterCreeper, 2023-04-01:
确认是1.8V，原来的是GD25LQ64ESIG，不知道处理器是否支持读取W25Q64FWSSIG（似乎指令有一点点差异）

MarsDoge, 2023-04-01:
> 确认是1.8V，原来的是GD25LQ64ESIG，不知道处理器是否支持读取W25Q64FWSSIG（似乎指令有一点点差异）

如果是与原版flash电压相同的情况下，W25Q64FWSSIG 是可以支持的.

leavelet, 2023-04-01:
我们使用了该版本的固件，能正常自检，但是没有显示输出，不知道是为啥

AnterCreeper, 2023-04-01:
Picture Refreshed:
[image]

AnterCreeper, 2023-04-01:
目前只能开机听到蜂鸣器一声，键盘大小写无反应，视频无信号输出
移除内存条，可以听见蜂鸣器持续报警

MarsDoge, 2023-04-01:
> 目前只能开机听到蜂鸣器一声，键盘大小写无反应，视频无信号输出 移除内存条，可以听见蜂鸣器持续报警

移除内存条报警有变化说明可以正常自检，第一次烧录开机需要多等一会儿，大概5分钟左右，再看看现象是否一致。

leavelet, 2023-04-09:
已成功点亮并稳定使用，感谢！