loongson/Firmware #46: 天创者L5A2主板更新BIOS不成功

toshcn, 2023-06-17:
天创者L5A2主板更新主板厂家给的版本：LS3A50007A-6-1-pason.fd后显示的版本号是：Loongs0n-UDK2018-U4.0.05420-stable202302
然后下载本仓库UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05429-stable202302_rel版本固件升级后不成功，bios版本仍然是显示厂家的版本，spi -u命令是显示两行ok的

MarsDoge, 2023-06-17:
> 天创者L5A2主板更新主板厂家给的版本：LS3A50007A-6-1-pason.fd后显示的版本号是：Loongs0n-UDK2018-U4.0.05420-stable202302 然后下载本仓库UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05429-stable202302_rel版本固件升级后不成功，bios版本仍然是显示厂家的版本，spi -u命令是显示两行ok的

你可以对比shell下的vers命令可以通过Release Time就可以查看出是否升级成功，你有可能已经升级成功了. 
目前不确定你这个LS3A50007A-6-1-pason.fd版本是从哪里获取的，但是有可能是同一基线所以界面显示为同一版本.

toshcn, 2023-06-17:
Build Time: 03/13/23 09:32:09
commit 182de671df164cb56b38e4b1a5296a004c159cac  Author: mengxiangdong <mengxiangdong@loongson.cn> Date: Fri Mar 3 16:13:29 2023 
不知是不是更新成功

toshcn, 2023-06-17:
LS3A50007A-6-1-pason.fd主板厂家给的

MarsDoge, 2023-06-17:
> Build Time: 03/13/23 09:32:09 commit 182de671df164cb56b38e4b1a5296a004c159cac Author: mengxiangdong [mengxiangdong@loongson.cn](mailto:mengxiangdong@loongson.cn) Date: Fri Mar 3 16:13:29 2023 不知是不是更新成功

从这个信息来看的确没有更新成功，我还是怀疑你可能更新的流程存在问题, 但这可能并不重要, 厂商提供的这个commit版本应该也可以正常使用，你目前是否有遇到此问题以外的问题吗? 如若没有建议你可以使用厂商提供的版本或者再次按更新流程尝试一下.

toshcn, 2023-06-17:
厂家的版本无法引导安装新世界的archlinux

closed 2023-11-30 by MarsDoge