loongson/Firmware #51: 7a2000 evb iommu初始化失败

FanFansfan, 2023-07-23:
`
[shangyatsen@loongson ~]$ dmesg | grep iommu
[    0.000000] Kernel command line: BOOT_IMAGE=/vmlinuz-6.4.1-g7bb3d7287fb4-dirty root=/dev/nvme0n1p2 rw rootflags=subvol=@ rootfstype=btrfs console=ttyS0,115200 loglevel=8 loongson_iommu=on
[    0.000000] Unknown kernel command line parameters "BOOT_IMAGE=/vmlinuz-6.4.1-g7bb3d7287fb4-dirty loongson_iommu=on", will be passed to user space.
[    0.365611] iommu: Default domain type: Translated
[    0.365614] iommu: DMA domain TLB invalidation policy: strict mode
[    3.462911] loongson_iommu_ivrs_init get ivrs table failed
[    3.468360] Failed to init iommu by ivrs
[    3.472718] loongson-iommu 0000:00:1a.0: can't derive routing for PCI INT A
[    3.479638] loongson-iommu 0000:00:1a.0: PCI INT A: no GSI
[    3.485099] iommu membase 00000000da1d5c61 pgtbase 00000000f79661b1 pgtsize 4000000 maxpages 1000
[   13.559937]     loongson_iommu=on
`
如日志显示，报错。/sys/kernel/iommu_groups/下全空

MarsDoge, 2023-08-04:
> 如日志显示，报错。/sys/kernel/iommu_groups/下全空

5月份及其之后的版本会支持iommu的ivrs, 5月份及其之前的版本不建议界面开启iommu.

FanFansfan, 2023-08-06:
> 

> 5月份及其之后的版本会支持iommu的ivrs, 5月份及其之前的版本不建议界面开启iommu.

最新的是三月版本。。通过efi var开启就能用吗？

closed 2023-11-04 by kilaterlee

FanFansfan, 2023-11-07:
chenhuacai/linux/issues/#2
上游内核维护者认为prestable202308固件没有传递正确的中断信息