loongson/Firmware #54: 天创者L5A2主板更新BIOS后启动失败

libreliu, 2023-07-27:
主板上面的丝印是 `LS3A5000_7A2000_EVB_BOARD_V1.5`，原来的固件在 UEFI Shell 的 vers 信息如下：

```
Version: Loongson UEFI (3A50007A2000_Desktop) V2.0.05269-beta8
Build Time: 2022 09 16 10:48:49 CST
Commit: commit ac658fe4df9a42a46931c893341f118a71ee4f53 Author: mengxiangdong <mengxiangdong@loongson.cn> Date: Thu Sep 15 14:48:54 2022 +0800
```

用 `spi -u UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05429-stable202302_dbg.fd` 刷入后，重启就没有任何显示输出了。

想请问：
1. 可能是什么样的问题呢..？
2. 固件上有什么调试的办法吗，是否可以确定一下启动不起来的具体原因？（UART..？）

MarsDoge, 2023-07-27:
@libreliu 第一次刷新需要开机少等5分钟左右做显存训练，下一版本已经优化.

libreliu, 2023-07-27:
> @libreliu bios更新后第一次启动需要开机稍等5分钟左右做显存训练，下一版本已经优化.

谢谢，等了五分钟还是不行，不过参考 #35 里面，拔掉一条内存之后发现可以启动了。

原来的固件是可以用 2x32G 内存的，现在的固件似乎同时插上就不能点亮了。这个问题有相关的排查方法嘛？

libreliu, 2023-07-27:
现在又可以啦...（不知道发生了什么）

不过谢谢回复！

closed 2023-07-27 by libreliu

xry111, 2023-07-27:
> 现在又可以啦...（不知道发生了什么）
> 
> 不过谢谢回复！

[离题的吐槽]

我记得以前玩 3A4000 的时候经常发生这种鬼事，机器如果几天不开就得把内存重插一遍……

反正大概来说确实是内存越大兼容性越差，我新买的 13900K 装 64 GB 内存连 XMP 都开不了 :(