loongson/Firmware #56: 无法在新世界固件的 A2101 主板和 L71 笔记本上启动 AOSC OS LiveKit

MingcongBai, 2023-08-11:
AOSC OS 的 LiveKit 启动盘在搭载新世界固件的龙梦 A2101 主板和同方 L71 笔记本上启动失败，在 EFI Shell 中发现 bootloongarch64.efi 文件被截断为了 bootloon.efi 的格式。启动盘是用 grub-mkrescue 生成的，但在其他架构上没有发现这个问题（亦有可能是因为其他架构少有超过 8.3 格式的 .EFI 可执行文件名）。

有问题的 iso 可以从这里下载 https://releases.aosc.io/os-loongarch64/livekit/aosc-os_livekit_20230707_loongarch64.iso 。

此外，Qemu 8.0 中并不能复现这个问题。

注：笔记本机主为 @KatyushaScarlet

xry111, 2023-08-11:
我有点怀疑 AOSC 的启动盘的 ESP 是不是 FAT16……

KatyushaScarlet, 2023-08-11:
以下是在L71主板上的情况：
[image]
ISO写入USB设备的方式为dd

xry111, 2023-08-11:
嘶，AOSC 的 ISO 里面 ESP 是个 FAT**12** 分区：

/dev/loop0p2: DOS/MBR boot sector, code offset 0x3c+2, OEM-ID "MTOO4042", sectors/cluster 2, root entries 240, sectors 5760 (volumes <=32 MB), sectors/FAT 9, sectors/track 36, serial number 0x2303281d, unlabeled, FAT (12 bit), followed by FAT

xry111, 2023-08-11:
把 ESP 弄成 FAT16，顺便把路径改成全大写试一下？

我现在只能远程访问我的 LoongArch 设备所以没法自己试：

```
losetup /dev/loop0 aosc-os_livekit_20230707_loongarch64.iso -P
mount -t vfat /dev/loop0p2 /mnt
cp /mnt/efi/boot/bootloongarch64.efi /tmp/BOOTLOONGARCH64.efi
umount /mnt
mkfs.vfat -F 16 /dev/loop0p2
mount -t vfat /dev/loop0p2 /mnt
mkdir /mnt/EFI/BOOT
cp /tmp/BOOTLOONGARCH64.efi /mnt/EFI/BOOT
umount /mnt
losetup -d /dev/loop0
```

MingcongBai, 2023-08-11:
我手上目前也没有可以直接用的设备，或许可以等下周让 @eatradish 回家试试看……

另供参考：这是我们的 [grub-mkrescue 参数](https://github.com/AOSC-Dev/aosc-mklive/blob/master/aosc-mklive.sh#L135)

jiegec, 2023-08-12:
但即使是 FAT12，只要能识别 LFN，文件名就是完整的。我测了一下，在 Linux 下面 mount AOSC 的 image，是可以看到完整的文件名的，说明是有 LFN。

xry111, 2023-08-12:
> 但即使是 FAT12，只要能识别 LFN，文件名就是完整的。我测了一下，在 Linux 下面 mount AOSC 的 image，是可以看到完整的文件名的，说明是有 LFN。
> 
> 用 QEMU+EDKII 上游固件测了一下，确实能看到完整的文件名，也可以自动启动到 LiveOS。

是的，所以这可能是个 bug，但是如果 FAT16 就能用的话可以先用着。

yetist, 2023-08-12:
> AOSC OS 的 LiveKit 启动盘在搭载新世界固件的龙梦 A2101 主板和同方 L71 笔记本上启动失败，在 EFI Shell 中发现 bootloongarch64.efi 文件被截断为了 bootloon.efi 的格式。启动盘是用 grub-mkrescue 生成的，但在其他架构上没有发现这个问题（亦有可能是因为其他架构少有超过 8.3 格式的 .EFI 可执行文件名）。 [...]

应该是iso制作有问题

jiegec, 2023-08-12:
对比 AOSC OS 和 LoongArchLinux：

```
NAME        FSTYPE      FSVER            LABEL       UUID                                   FSAVAIL FSUSE% MOUNTPOINTS
loop0       iso9660                      LiveKit     2023-07-07-00-53-58-00
├─loop0p1
├─loop0p2   vfat        FAT12                        2303-281D
└─loop0p3
loop1       iso9660     Joliet Extension ARCH_202308 2023-08-04-04-53-04-00
├─loop1p1   iso9660     Joliet Extension ARCH_202308 2023-08-04-04-53-04-00
├─loop1p2   vfat        FAT16            ARCHISO_EFI 501C-4C7F
└─loop1p3
```

tivycc, 2023-08-14:
> AOSC OS 的 LiveKit 启动盘在搭载新世界固件的龙梦 A2101 主板和同方 L71 笔记本上启动失败，在 EFI Shell 中发现 bootloongarch64.efi 文件被截断为了 bootloon.efi 的格式。启动盘是用 grub-mkrescue 生成的，但在其他架构上没有发现这个问题（亦有可能是因为其他架构少有超过 8.3 格式的 .EFI 可执行文件名）。 [...]

1、采用商业版昆仑固件V4.0，可以准确识别文件名。
2、移植商业版文件系统至开源BIOS，仍可复现
可以公布EFI分区制作的详细步骤，与此同时追踪BIOS识别过程

MingcongBai, 2023-09-13:
另外一个案例，TC512A0 主板也有同样问题

MingcongBai, 2023-09-20:
终于破案，是由于 GRUB 中的 `util/grub-mkrescue.c` 没有指定 ISO Level 导致的；工具似乎默认选用了 ISO Level 1，文件名限制为 8.3 长度了。给 GRUB 打如下补丁后生成的 ISO，可以在 A2101/TC512A0 上启动了：

```
diff -Naur grub-2.12-rc1/util/grub-mkrescue.c grub-2.12-rc1.iso-level/util/grub-mkrescue.c
--- grub-2.12-rc1/util/grub-mkrescue.c  2023-09-19 07:56:32.823829975 +0000
+++ grub-2.12-rc1.iso-level/util/grub-mkrescue.c        2023-09-19 18:09:26.979596573 +0000
@@ -514,6 +514,8 @@
   xorriso_push ("-as");
   xorriso_push ("mkisofs");
   xorriso_push ("-graft-points");
+  xorriso_push ("-iso-level");
+  xorriso_push ("3");

   iso9660_dir = grub_util_make_temporary_dir ();
   grub_util_info ("temporary iso9660 dir is `%s'", iso9660_dir);
```

MingcongBai, 2023-09-20:
GRUB 修复已[提交到上游](https://lists.gnu.org/archive/html/grub-devel/2023-09/msg00069.html)，希望能赶上 2.12.

MarsDoge, 2024-09-14:
@MingcongBai 确认该issues是否可以关闭？

MingcongBai, 2024-09-14:
是的，可以关闭了，感谢提醒 @MarsDoge

closed 2024-09-14 by MingcongBai