loongson/Firmware #61: 刷新 UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05429-stable202302_rel.fd 成功后，U盘安装 Loongnix-20.5.livecd.loongarch64.iso 花屏

dingymm, 2023-08-30:
刷新 UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05429-stable202302_rel.fd 成功后，U盘安装 Loongnix-20.5.livecd.loongarch64.iso 花屏

选择 安装选项后，打开图形界面时，花屏，顶部几行黄色，屏幕最下面4分之1处显示 白色花屏

dingymm, 2023-08-30:
[image]

MarsDoge, 2023-08-30:
@dingymm 请验证https://github.com/loongson/Firmware/blob/3e33268a0bc86f39715141304e2a112415fb3b3b/5000Series/PC/3A5000-7A2000-EVB/UDK2018_3A5000-7A2000_Desktop_EVB_V4.0.05536-Prestable202308_dbg.fd

dingymm, 2023-08-30:
感谢回复~已经刷新好了。内容大更新啊！
[image]
但是引导安装程序还是
[image]
其他的情况，
1、我没有安装独立显卡，2、内存已经更换了几种，目前是8g单根的英睿达，插在靠近cpu的位置，3，u盘是用rufus写入的。

dingymm, 2023-08-30:
问题找到了！~！！我分别用hdmi在两台显示器dell的2417 1920x1200 和 飞利浦2K 都是一样的问题。然后找出一台 优派va2232 19寸的 vga显示器 用vga连接，可以显示出画面了。

dingymm, 2023-08-30:
后面我又测试，用HDMI口，接这台可以用的优派va2232，一样花屏。所以怀疑是主板HDMI接口输出部分的问题。希望官方可以解决。

closed 2023-08-31 by dingymm