loongson/Firmware #64: 天创者

lshw, 2023-09-12:


lshw, 2023-09-12:
flashrom程序用CH341A_SPI 写新的rom, 启动时南桥风扇不转，串口无输出
刷回原来的ROM,南桥风扇可以转动， 但屏幕黑屏，串口输出如下， 请问我该怎么办？
                                                                                                                                                             
base 1518: 0000000000000000                                                                                                                                                                    
mask 1598: 0000000000000000                                                                                                                                                                    
mmap 1618: 0000000000000000                                                                                                                                                                    
base 1520: 0000000000000000                                                                                                                                                                    
mask 15a0: 0000000000000000                                                                                                                                                                    
mmap 1620: 0000000000000000                                                                                                                                                                    
base 1528: 0000000000000000                                                                                                                                                                    
mask 15a8: 0000000000000000                                                                                                                                                                    
mmap 1628: 0000000000000000                                                                                                                                                                    
base 1530: 0000000000000000                                                                                                                                                                    
mask 15b0: 0000000000000000                                                                                                                                                                    
mmap 1630: 0000000000000000                                                                                                                                                                    
base 1538: 0000000000000000                                                                                                                                                                    
mask 15b8: 0000000000000000                                                                                                                                                                    
mmap 1638: 0000000000000000                                                                                                                                                                    
Gmem init done!                                                                                                                                                                                
syn0 Mac is valid, Mac address:                                                                                                                                                                
38:F7:CD:C4:2C:9F                                                                                                                                                                              
Error::: PHY not responding Busy bit dIdnot get cleared !!!!!!                                                                                                                                 
Error::: PHY not responding Busy bit dIdnot get cleared !!!!!!                                                                                                                                 
Error::: PHY not responding Busy bit dIdnot get cleared !!!!!!                                                                                                                                 
Error::: PHY not responding Busy bit dIdnot get cleared !!!!!!                                                                                                                                 
syn1 Mac is invalid, Mac address:                                                                                                                                                              
FF:FF:FF:FF:FF:FF                                                                                                                                                                              
now get a new mac                                                                                                                                                                              
syn1 New Mac address:                                                                                                                                                                          
0E:99:66:39:CF:C8                                                                                                                                                                              
[... cut ...]
                                                                                                                                                                                               
ASSERT_EFI_ERROR (Status = Not Found)                                                                                                                                                          
ASSERT [DxeIpl] /home/mengxiangdong/new/UEFI/uefi-sz/MdeModulePkg/Core/DxeIplPeim/DxeLoad.c(469): !EFI_ERROR (Status)

MarsDoge, 2023-09-13:
rom烧写不完整，需要去确定烧录的电平是3.3还是1.8v.

lshw, 2023-09-14:
确实是rom烧写不完整，我原来备份的rom不对，
不过也发现这里的fd文件， 直接用写片机写到rom里， 是不能启动的。

closed 2023-09-14 by lshw

lshw, 2023-09-14:
刷了个4.0.5569-stable202308的rom ， 好了

lshw, 2023-09-14:
4.0.05569不行，还是黑屏，
4.0.05536可以了。