loongson/Firmware #66: 天创者L5A2 V1.5主板，新bios偶尔启动不成功

lshw, 2023-09-15:
不成功时，显示如下信息:
============================================
Pages Offset = 0x6EE00000
Loading PEIM at 0x000FEFE8000 EntryPoint=0x000FEFE8240 PeiCore.efi
Reinstall PPI: 8C8CE578-8A3D-4F1C-9935-896185C32DD3
Reinstall PPI: 5473C07A-3DCB-4DCA-BD6F-1E9689E7349A
Reinstall PPI: B9E0ABFE-5979-4914-977F-6DEE78C278A6
Install PPI: F894643D-C449-42D1-8EA8-85BDD8C65BDE
Loading PEIM at 0x000FEFDC000 EntryPoint=0x000FEFDC240 DxeIpl.efi
PROGRESS CODE: V03020002 I0
PROGRESS CODE: V03020003 I0
Install PPI: EE4E5898-3914-4259-9D6E-DC7BD79403CF
Install PPI: 1A36E4E7-FAB6-476A-8E75-695A0576FDD7
Install PPI: 0AE8CE5D-E448-4437-A8D7-EBF5F194F731
Customized Guided section Memory Size required is 0xB50010 and address is 0xFE47B000
Extract guided section Failed - Invalid Parameter
Customized Guided section Memory Size required is 0xB50010 and address is 0xFD91A000
Extract guided section Failed - Invalid Parameter
Customized Guided section Memory Size required is 0xB50010 and address is 0xFCDB9000
Extract guided section Failed - Invalid Parameter
Customized Guided section Memory Size required is 0xB50010 and address is 0xFC258000
Extract guided section Failed - Invalid Parameter
DXE IPL Entry
PROGRESS CODE: V03031000 I0

ASSERT_EFI_ERROR (Status = Not Found)
ASSERT [DxeIpl] /data/qiandongyan/push/zuixin/push2/rd/uefi-loongarch-clean_master-test/MdeModulePkg/Core/DxeIplPeim/DxeLoad.c(469): !EFI_ERROR (Status)
=================================================================

成功时显示如下信息
=================================================================
TemporaryRamMigration(0x90100000, 0xC0440000, 0x300000)
        NewHeap created successfully, NewHeap Range: (0xC0640000 - 0xC0740000)
        NewStack created successfully, NewStack Range: (0xC0440000 - 0xC0640000)
        Sp OldPoint 0x9027F878, NewPoint 0xC04BF878
Pages Offset = 0x6EE00000
Loading PEIM at 0x000FEFE8000 EntryPoint=0x000FEFE8240 PeiCore.efi
Reinstall PPI: 8C8CE578-8A3D-4F1C-9935-896185C32DD3
Reinstall PPI: 5473C07A-3DCB-4DCA-BD6F-1E9689E7349A
Reinstall PPI: B9E0ABFE-5979-4914-977F-6DEE78C278A6
Install PPI: F894643D-C449-42D1-8EA8-85BDD8C65BDE
Loading PEIM at 0x000FEFDC000 EntryPoint=0x000FEFDC240 DxeIpl.efi
PROGRESS CODE: V03020002 I0
PROGRESS CODE: V03020003 I0
Install PPI: EE4E5898-3914-4259-9D6E-DC7BD79403CF
Install PPI: 1A36E4E7-FAB6-476A-8E75-695A0576FDD7
Install PPI: 0AE8CE5D-E448-4437-A8D7-EBF5F194F731
Customized Guided section Memory Size required is 0xB50010 and address is 0xFE47B000
ProcessFvFile() FV at 0xFE47B010, FvAlignment required is 0x10
Install PPI: EA7CA24B-DED5-4DAD-A389-BF827E8F9B38
Notify: PPI Guid: EA7CA24B-DED5-4DAD-A389-BF827E8F9B38, Peim notify entry point: 90026BA8
The 2th FV start address is 0x000FE47B010, size is 0x00B50000, handle is 0xFE47B010
Install PPI: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
Notify: PPI Guid: 49EDB1C1-BF21-4761-BB12-EB0031AABB39, Peim notify entry point: 90026BA8
The Fv FE47B010 has already been processed!
DXE IPL Entry
Loading PEIM at 0x000FE441000 EntryPoint=0x000FE441240 DxeCore.efi
PROGRESS CODE: V03021001 I0
Loading DXE CORE at 0x000FE441000 EntryPoint=0x000FE441240
Install PPI: 605EA650-C65C-42E1-BA80-91A52AB618C6
==============================================================================

MarsDoge, 2023-09-15:
初步看是flash芯片不稳定或者还是烧录文件信息不全.

lshw, 2023-09-22:
嗯， 用网线里的单股，把座绑了一下， 没在出现问题， 可能是启动是接触是好的， 但是启动过程中接触不良，导致后面的过程中出现读取错误。