loongson/Firmware #74: 3A6000 (XA61200 V1_1) fails to boot if RDNA graphics card is present (PciBusDxe error)

Civil, 2023-11-16:
Firmware version: 4.0.05569-stable202308

If RDNA or newer graphics card is in the system, system doesn't boot at all, stuck repeating:
```
ERROR: C40000002:V02010007 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FC198698
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FC198C18
```

Which seems to originate from PciBusDxe.efi

I've tried that with RX6400 (RDNA 2) and with RX 5700 XT (RNDA) with same result.

Full boot log:
[loongson_boot_stuck.log](https://github.com/loongson/Firmware/files/13376975/loongson_boot_stuck.log)

jiegec, 2023-11-16:
Same problem here (XA612A0_V1.1, AMD RX6400):

```
InstallProtocolInterface: 387477C1-69C7-11D2-8E39-00A0C969723B FCA232C0
InstallProtocolInterface: DD9E7534-7762-4698-8C14-F58517A625AA FCA233A8
InstallProtocolInterface: 387477C2-69C7-11D2-8E39-00A0C969723B FCA232D8
InstallProtocolInterface: 09576E91-6D3F-11D2-8E39-00A0C969723B FCABC918
InstallProtocolInterface: D3B36F2B-D551-11D4-9A46-0090273FC14D 0
InstallProtocolInterface: D3B36F2C-D551-11D4-9A46-0090273FC14D 0
InstallProtocolInterface: D3B36F2D-D551-11D4-9A46-0090273FC14D 0
GraphicsConsole video resolution 640 x 480
Graphics - Mode 0, Column = 80, Row = 25
Graphics - Mode 1, Column = 0, Row = 0
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
ERROR: C40000002:V02010008 I0 93B80004-9FB3-11D4-9A3A-0090273FC14D FCA21098
```

Full log: [boot-AMD-RX6400.log](https://github.com/loongson/Firmware/files/13377749/boot-AMD-RX6400.log)

ruankaijian, 2023-11-29:
可以刷[LS3A60007A-11-23-bootorder.fd](https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6/LS3A60007A-11-23-bootorder.fd)固件测试一下：刷了固件后，在bios里面把4G大空间打开

jiegec, 2023-11-29:
> 可以刷[LS3A60007A-11-23-bootorder.fd](https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6/LS3A60007A-11-23-bootorder.fd)固件测试一下：刷了固件后，在bios里面把4G大空间打开

请问这个固件可以用于 XA612A0 吗（LS3A6000-7A2000-1w-EVB-V1.21）？

ruankaijian, 2023-11-29:
可以

ruankaijian, 2023-11-29:
上面问题已知，AMD的高端显卡都起不来，目前新版本正式版的固件最近发布，等新固件发布会修复这个问题。

ruankaijian, 2023-11-29:
您们方便的时候麻烦提供一下显卡的vid和did，我单独出一个版本给您们

jiegec, 2023-11-29:
很遗憾，升级 LS3A60007A-11-23-bootorder.fd 以后，AMD RX6400 还是起不来：

[boot-new-firmware-AMD-RX6400.log](https://github.com/loongson/Firmware/files/13497410/boot-new-firmware-AMD-RX6400.log)

在 XA612A0 上的操作记录：
1. 升级前固件版本（dmidecode）：Loongson-UDK2018-V4.0.05494-stable202305
2. 在uefi shell跑spi -u LS3A60007A-11-23-bootorder.fd 固件
3. 升级后固件版本（dmidecode）：Loongson-UDK2018-V4.0.05569-stable202308，观察到 dmidecode 输出了DRAM的频率等信息，这是旧版本所没有的
4. 拆下 AMD RX550，换上 AMD RX6400后，固件初始化失败
5. 拆下 AMD RX6400，换回 AMD RX550，固件初始化成功，可以正常显示

jiegec, 2023-11-29:
> 您们方便的时候麻烦提供一下显卡的vid和did，我单独出一个版本给您们

我手上没有其他的主机，看不到 vid 和 pid，但是网上可以查到一个：1002:743f

ruankaijian, 2023-11-29:
我先以你提供的id给你提供一版固件

ruankaijian, 2023-11-29:
AMD RX6400 定制固件版本[LS3A60007A-11-29-skip743F.fd](https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6-RX6400/LS3A60007A-11-29-skip743F.fd)

ruankaijian, 2023-11-29:
> > 方便各位的时候麻烦提供一下显卡的视频，我单独出一个版本给你们
> 
> 我手上没有其他的主机，但是看不到vid和pid，网上可以查到一个：1002:743f
> 
> PS 有群友（@xm_moe）表示，在 16x 的 PCIe 插槽插一个旧的 AMD 显卡（如 RX550），再在离 CPU 更远的 8x 的 PCIe 插槽插一个新的 AMD 显卡（如 RX6400，需要解决 x8）槽插x16卡的问题），可以绕过这个问题（让固件去初始化RX550而不是去管RX6400），但是我这里PCIe插槽已经用来插无线网卡了，有点不好实现。

固件下载链接：https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6-RX6400/LS3A60007A-11-29-skip.fd

jiegec, 2023-11-29:
> 固件下载链接：https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6-RX6400/LS3A60007A-11-29-skip.fd

非常感谢！看起来 #79 的版本更加通用一些，我先试试 #79 的固件版本，如果还是不行再来试试这个。

jiegec, 2023-11-29:
> AMD RX6400 定制固件版本[LS3A60007A-11-29-skip743F.fd](https://gitea.whlug.cn/3A6000/3A6000/releases/download/XA61200%E4%B8%BB%E6%9D%BF%E5%9B%BA%E4%BB%B6-RX6400/LS3A60007A-11-29-skip743F.fd)

感谢 @ruankaijian ！我已经用 #79 的固件解决了问题，就没有再去测这个版本。

closed 2023-11-30 by kilaterlee