loongson/Firmware #82: SMBIOS Type 44 ISA Extensions Support 字段实现存疑

jiegec, 2023-12-11:
SMBIOS Type 44 在 [3.3 LoongArch Processor-specific Block Structure](https://loongson.github.io/LoongArch-Documentation/LoongArch-Processor-SMBIOS-Spec-EN.html#_3_3_loongarch_processor_specific_block_structure) 中定义 ISA Extensions Support 字段：

- Offset: 24h
- Description: The bit field [3:0] indicates support for the existing LoongArch standard ISA extensions. It is modeled after the LoongArch EUEN register (CSR 0x2), and meaning of each bit is the same as defined for the EUEN register.
Setting a bit in this field indicates that this system supports the corresponding ISA extension.

从 3C5000 导出的数据：

```
Handle 0x0015, DMI type 44, 48 bytes
Unknown Type
        Header and Data:
                2C 30 15 00 01 00 28 0A 00 01 28 00 4C 6F 6F 6E
                67 73 6F 6E 00 00 00 00 00 00 00 00 33 43 35 30
                30 30 00 00 00 00 00 00 00 00 00 00 C7 CF 7C 00
        Strings:
                ptal....X
```

ISA Extensions Support 字段的值为 0x007CCFC7。

从 3A6000 导出的数据：

```
Handle 0x001C, DMI type 44, 48 bytes
Unknown Type
        Header and Data:
                2C 30 1C 00 01 00 28 0A 00 01 28 00 4C 6F 6F 6E
                67 73 6F 6E 00 00 00 00 00 00 00 00 33 41 36 30
                30 30 2D 48 56 00 00 00 00 00 00 00 C7 CC 7C 7E
        Strings:
                ptal....X
```

ISA Extensions Support 字段的值为 0x7E7CCCC7。

按照标准，最低四位记录了是否支持扩展：FP，LSX，LASX，LBT。但是问题来了：

1. 上面的两个例子里，最低四位都等于 0x7，表示支持 FP、LSX 和 LASX，不支持 LBT，和实际不符合
2. 其余的位数代表什么含义？没有看到定义。

背景：https://github.com/loongson-community/discussions/issues/28

jiegec, 2023-12-11:
发现 ISA Extensions Support 字段的值被填成了 CPUCFG[1]：

- 3A6000：0x7e7cccc7
- 3C5000：0x007ccfc7

和前面的内容是一致的。那么还是上面的问题，是固件实现错了，还是 SMBIOS 定义过时了？

MarsDoge, 2026-01-12:
以RefCode stable2511为基准进行验证Type44.ISA Extensions Support: 
参考 @jiegec https://github.com/jiegec/dmidecode/commit/3ade507bc796e548a8f27eb5f782160f16b648e9 解析如下：

```
3A6000：
Handle 0x001C, DMI type 44, 48 bytes
Processor Additional Information
        Referenced Handle: 0x0001
        Processor Type: 64-bit LoongArch (LoongArch64)
        Machine Vendor ID: Loongson
        CPU ID: 3A6000-HV
        ISA Extensions Support:
                Floating Point Instructions
                128-bit Vector Instructions
                256-bit Vector Instructions
                Binary Translation Instructions

3C6000：
Handle 0x007D, DMI type 44, 48 bytes
Processor Additional Information
        Referenced Handle: 0x0002
        Processor Type: 64-bit LoongArch (LoongArch64)
        Machine Vendor ID: Loongson
        CPU ID: 3C6000/D
        ISA Extensions Support:
                Floating Point Instructions
                128-bit Vector Instructions
                256-bit Vector Instructions
                Binary Translation Instructions
```
Type44.ISA Extensions Support 数据来自 CPUCFG[2] .

closed 2026-04-02 by MarsDoge