loongson/Firmware #84: LS2C5LE: 进入系统时触发 kernel panic, 导致无法进入系统

KFERMercer, 2023-12-28:
问题描述: 从 iso 启动时触发 kernel panic, 无法进入系统安装. (由于无法成功安装系统, 无法确认启动本地系统时是否触发)

主板型号: [LS2C5LE](https://github.com/loongson/Firmware/raw/main/Image/LS2C5LE.jpg) (双路 3C5000L 7A1000)

[image]

bios版本: [`Loongson-UDK2018-V2.0.0-prebeta9`](https://github.com/loongson/Firmware/blob/main/5000Series/Server/LS2C5LE/UDK2018-LoongArch-LS2C5LE-pre-beta9.fd)

内存条安装插槽: A1 B1 A3 B3 各一条紫光国芯 16g 3200

测试iso: 

1. `archlinux-livecd-2023.11.29-loong64.iso`
2. `debian12-live-gnome-loong64-20230629.iso`
3. `proxmox-ve_8.1-3-loong64.iso`

内核日志：
```log
[    0.000000] Kernel panic - not syncing: Failed to allocate 481111834624 bytes for node 0 memory map
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 6.4.0-0-loong64 #1  Debian 6.4.0-1
[    0.000000] Hardware name: LOONGSON LOONGSON T100 T2208A/Loongson-LS2C5LE, BIOS Loongson-UDK2018-V2.0.0-prebeta9 11/01/2022
[    0.000000] Stack : 9000000001067bb8 0000000000000000 90000000002241f8 9000000001064000
[    0.000000]         9000000001067ad0 9000000001067ad8 0000000000000000 0000000000000000
[    0.000000]         9000000001067ad8 0000000000000000 0000000000000000 0000000000000000
[    0.000000]         0000000000000001 9000000001067ad8 0000000000000000 0000000000000000
[    0.000000]         0000000000017fe8 0000000000000000 90000000010fb2d8 0000000000000000
[    0.000000]         0000000000000000 0000000000000000 0000000000000000 0000000000000080
[    0.000000]         0000000000000000 0000000000000000 9000000000f737a0 9000000001084000
[    0.000000]         0000007004800000 00000000007ff974 90000000011b29c0 00000001c0120000
[    0.000000]         0000000000000000 0000000000000000 9000000000224210 0000000000000000
[    0.000000]         00000000000000b0 0000000000000004 0000000000000000 0000000000070800
[    0.000000]         ...
[    0.000000] Call Trace:
[    0.000000] [<9000000000224210>] show_stack+0x64/0x188
[    0.000000] [<9000000000c5e720>] dump_stack_lvl+0x60/0x88
[    0.000000] [<9000000000c52100>] panic+0x13c/0x30c
[    0.000000] [<9000000000c9628c>] free_area_init+0x974/0xd5c
[    0.000000] [<9000000000c857a0>] paging_init+0x40/0x60
[    0.000000] [<9000000000c8089c>] start_kernel+0xa4/0x658
[    0.000000] [<9000000000c610b4>] kernel_entry+0xb4/0xb8
[    0.000000] 
[    0.000000] ---[ end Kernel panic - not syncing: Failed to allocate 481111834624 bytes for node 0 memory map ]---
```

后根据日志信息, 拔掉属于 node 0 的 A1 插槽内的内存后, 无法通过自检:

自检串口输出:
```
Shut down slave cores done!

UEFI-EDK2 loongson Initializing...

CPU Version: 00000043
CPU Type: 00000005
CPU SRAM: 0005000180158081
Node 00000000
N Voltage  write :
v ctrl end

N Voltage  read :
0000047a

P Voltage write :
io ctrl end

Node 00000004
N Voltage  write :
v ctrl end

N Voltage  read :
[... 253 lines cut ...]
Waiting node2 F HyperTransport bus to be up.
>0x44 20
Waiting node4 F HyperTransport bus to be up.
>0x44 20
Waiting node7 F HyperTransport bus to be up.
>0x44 20
Waiting node5 F HyperTransport bus to be up.
>0x44 20
Waiting node6 F HyperTransport bus to be up.
>0x44 20
LS3A-7A linkup.
FsbConnect Success...

***** MemConfigPpi Version v2.5 *****

erase flash 0x800000001c038000 for store mc bit map
Erase end!
check node 0 mc0 slot0
NO DIMM in this slot.
check node 0 mc0 slot1
NO DIMM in this slot.
final tCKmax 0
No dimm on node 0
start up other CPU
```

KFERMercer, 2023-12-30:
此故障实际上导致 大别山T2218A 服务器无法安装任何新世界系统.

MarsDoge, 2024-01-12:
@KFERMercer Node0节点不支持内存不在位，必须保留内存，在此基础上请确认是否在配置界面是否已关闭legacymode模式引导社区版系统.

KFERMercer, 2024-01-12:
@MarsDoge 感谢回应

> Node0节点不支持内存不在位，必须保留内存

后续尝试只插 A2 (也归属于node0) 依然无法加载内核, 故障同描述中 A1 插有内存的情况.

> 在此基础上请确认是否在配置界面是否已关闭legacymode模式引导社区版系统.

此版本固件无 legacymode 配置选项.

MarsDoge, 2024-01-13:
验证 [archlinux-livecd-2023.11.29-loong64.iso](https://mirrors.pku.edu.cn/loongarch/archlinux/iso/2023.11.29/archlinux-livecd-2023.11.29-loong64.iso) : 目前看似是最新的内核可能启用了某些配置，导致搭配旧一些的固件引导calltrace，请补充一下您内核的完整log. @KFERMercer 
或者您可以是否方便尝试一下6.1的内核.