loongson/Firmware #88: XA612A0: cannot restore from sleep

xen0n, 2024-01-04:
I just accidentally pressed the power button and the board went into ACPI sleep, but when I tried to wake it up it dies with the following serial console output:

```
Facs = F9C60000
Ra = 90000000033F8068
Jump to Kernel...

!!!! LoongArch64 Exception Type - 00(#INT - Interrupt(CSR.ECFG.VS=0)) !!!!
Zero  - 0x0000000000000000, RA  - 0x90000000033F8068, TP - 0xFBFFF7BFF7FFFFEF, SP - 0x0000000000000000
  A0  - 0x0000000000000013, A1  - 0x000000000000000A, A2 - 0x800000001FE001E0, A3 - 0x0000000000000000
  A4  - 0x0000000000000001, A5  - 0x0000000000000001, A6 - 0x0000000000000003, A7 - 0x0000000000000002
  T0  - 0x9000000003EF0478, T1  - 0x90000000033F8084, T2 - 0x0000000090400000, T3 - 0x0000000090400000
  T4  - 0x8000000000000000, T5  - 0x0000000000000000, T6 - 0x0000000000000002, T7 - 0x0000000000000002
  T8  - 0x0000000000000000, R21 - 0xFFFF77FFF7FFFFFF, FP - 0x0000000000000000, S0 - 0x90000000033F8068
  S1  - 0x00000000900B8378, S2  - 0x000000009027FAB8, S3 - 0x0000000000000000, S4 - 0x0000000000000000
  S5  - 0x80000EFDFE003018, S6  - 0x0000000090100248, S7 - 0x000000009027FFA0, S8 - 0x0000000000000020

CRMD  - 0x00000000000000A8, PRMD  - 0x0000000000000000, EUEN - 0x0000000000000000, MISC - 0x0000000000000000
ECFG  - 0x0000000000000000, ESTAT - 0x0000000000000000, ERA  - 0x0000000000000000, BADV - 0x0000000000000000
BADI  - 0x0000000000000000
!!!! Can't find image information. !!!!
```

This is UDK2018-LoongArch-V4.0.05132-beta10 (#86).

MarsDoge, 2024-08-22:
Hi, xen0n, maybe you can close this issue now.

xry111, 2024-08-23:
I can confirm it's fixed by 2405 update.

xen0n, 2024-08-23:
My XA612A0 box was upgraded to XC-LS3A6M (to make my bedroom quieter) so I couldn't verify the firmware update, but thanks for pinging the thread! As @xry111 has confirmed the fix I'll just close this issue then.

closed 2024-08-23 by xen0n