loongson/Firmware #91: 联想开天M549T-D015申请新固件

machuang666, 2024-02-19:
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: ZD-TECH
        Version: W05KT1DA
        Release Date: 04/14/2022
        ROM Size: 8192 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                Serial services are supported (int 14h)
                USB legacy is supported
                Function key-initiated network boot is supported
                UEFI is supported
        BIOS Revision: 4.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Lenovo
        Product Name: 90TWS008KX
        Version: KaiTianM549T-D015
        Serial Number: KT00FN36
        UUID: c8894fbe-6ede-46b2-a1a5-c29c39edc804
        Wake-up Type: Power Switch
        SKU Number: Loongson_SKU
        Family: Type1Family

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: Loongson
        Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
        Version: 1.0
        Serial Number: MX20BA001J
        Asset Tag: Type2 - Board Asset Tag
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Type2 - Board Chassis Location
        Chassis Handle: 0xFFFF
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
        Manufacturer: Loongson
        Type: Rack Mount Chassis
        Lock: Not Present
        Version: Version To Be Filled By O.E.M
        Serial Number: Chassis Board Serial#To Be Filled By O.E.M
        Asset Tag: Chassis Board Asset Tag#To Be Filled By O.E.M
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
        Security Status: None
        OEM Information: 0x00000000
        Height: Unspecified
        Number Of Power Cords: 1
        Contained Elements: 0
        SKU Number: SKU Number#To Be Filled By O.E.M

dg1vg4, 2024-09-04:
你的固件能分享一个么？
我的是开天M540z,固件版本还在2021年5月份。
说不定主板是同一个版型。

elysia-best, 2025-07-02:
我的主板是L5LMB01，使用的BIOS是这个

[Loongbios_LXKT_L5LMB01.zip](https://github.com/user-attachments/files/21021870/Loongbios_LXKT_L5LMB01.zip)

BIOS 版本：Loongson-UDK2018-U2.0.04011-beta2
显示的主板型号：Loongson-LS3A5000-7A1000-1w-v0.1-CRB 

[image]

希望对你们有用