loongson/Firmware #92: 2402固件r7 240显卡花屏

ziqi-cn, 2024-03-01:
XA61200主板刷入UDK2018-3A6000-7A2000_EVB_V4.0.05634_prestable2402rel.fd固件，r7 240显卡BIOS启动阶段花屏

acdcGGE, 2024-03-01:
[image]
[image]
rx550和rx 5500xt不会花屏  据群友测试大概只有r7 240 会花

MarsDoge, 2024-03-01:
帮忙提供一下花屏显卡的DeviceId @ziqi-cn

MarsDoge, 2024-03-02:
# AMDGraphicsDxe

UEFI AMD GOP DRIVER.

Currently supported GPUs and TBD GPUs:

| Family | Chipset name | Microarchitecture | ISA | Product name | Status |
| --- | --- | --- | --- | --- | --- |
| Southern Islands | CAPE VERDE, PITCAIRN, TAHITI, OLAND, HAINAN | GCN1.0+ | DCE 6.x | HD7750-HD7970, R9 270, R9 280, R9 370X, R7 240, R7 250 | Done |
| Sea Islands | BONAIRE, KABINI, KAVERI, HAWAII, MULLINS | GCN2.x | DCE 8.x | HD7790, R7 260, R9 290, R7 360, R9 390 | Done |
| Volcanic Islands | CARRIZO, FIJI, STONEY, TONGA, TOPAZ, WANI | GCN3.x | DCE 10/11.x | R9 285, R9 380, R9 380X, R9 Fury, R9 Nano, R9 Fury X, Pro Duo | Done |
| Arctic Islands | POLARIS10/11/12, VEGAM | GCN4.x | DCE 11.2 | RX 460, RX 470, RX 480, RX 540, RX 550, RX 560, RX 570, RX 580, RX 590, Pro WX 3200 | Done |
| Vega | VEGA10/11/12/20 | GCN5.x | DCE 12.x | RX Vega 56, RX Vega 64, Radeon Vega II, Radeon VII | TBD |
| Vega | RAVEN | GCN5.x | DCN 1.0 | Raven Ridge APU series | TBD |
| Vega | RENOIR | GCN5.x | DCN 2.1 | Renoir, Lucienne, and Cezanne APU series | TBD |
| Navi | NAVI10/14 | RDNA | DCN 2.0 | RX 5500, RX 5500 XT, RX 5600, RX 5600 XT, RX 5700, RX 5700 XT | TBD |
| Navi | NAVI21/22/23/24 | RDNA | DCN 3.0 | RX 6500 XT, RX 6600, RX 6600 XT, RX 6650 XT, RX 6700, RX 6700 XT, RX 6750 XT, RX 6800, RX 6800 XT, RX 6900 XT, RX 6950 XT | TBD |

感谢各位反馈，上述为2402目前Firmware Stage计划支持的显卡列表Status，其中根据 @ziqi-cn 同学反馈的情况，我们对同设备ID的R7 240进行了复测，发现不同厂商做的R7 240表现存在差异，我们会尽快调试该差异化，由于设备数量限制，我们会继续完善，非常感谢各位协助与配合！

MarsDoge, 2024-03-04:
@ziqi-cn 目前验证两种R7 240均正常，有时间请帮忙验证一下您的R7 240， [6000Series/PC/XA61200/UDK2018-3A6000-7A2000_EVB_V4.0.05634_prestable2402_0304dbg.fd](https://github.com/MarsDoge/Firmware/commit/fe48bb6bf9424dd0e26b989fca25fd560dcc457a#diff-787a6619b085d2ffd649e75c170d039671017d773fd729a88292d86dffcf4fe6) ，非常感谢！

MarsDoge, 2024-03-04:
@ziqi-cn 能够帮忙提供一下串口信息以及卡的照片先看一下

ziqi-cn, 2024-03-04:
> @ziqi-cn 能够帮忙提供一下串口信息以及卡的照片先看一下

我手上没有串口调试器，这是显卡照片

[image]

[image]

[image]

closed 2024-04-08 by ziqi-cn