loongson/Firmware #93: sd-boot 无法执行

hanyuwei70, 2024-03-02:
串口日志如下：
[console-2024-03-02-220502.log](https://github.com/loongson/Firmware/files/14469723/console-2024-03-02-220502.log)
BIOS Version: Loongson-UDK2018-V4.0.05636-stable202311

xry111, 2024-03-02:
Not a firmware bug.

https://github.com/loongson-community/discussions/issues/19

hanyuwei70, 2024-03-02:
> Not a firmware bug.
> 
> [loongson-community/discussions#19](https://github.com/loongson-community/discussions/issues/19)

可以绕过嘛？之前是可以正常启动的。

xry111, 2024-03-02:
AFAIK using LSX instructions in UEFI applications has never been supported.  Not sure what you mean by "before."  Maybe something has changed in systemd-boot or the toolchain (causing it to start generating LSX instructions).

yetist, 2024-04-09:
> > 参考这个补丁：[systemd/systemd#31608](https://github.com/systemd/systemd/pull/31608)
> 
> 如果发行版编译 GCC 的时候用了 `--with-arch=la464` 或者 `--with-simd=lsx` 这样的东西，libgcc.a 可能包含 SIMD 指令，而这个补丁无法解决该问题。
> 
> [loongson-community/discussions#19 (comment)](https://github.com/loongson-community/discussions/issues/19#issuecomment-1851720891)

这种情况确实没有办法，这个补丁只对编译 GCC 时未开启 lsx，但在编包时使用 CFLAGS=" -mlsx" 环境变量的情况有效。

jiegec, 2024-04-09:
那么要搞多个版本的 libgcc.a 吗，例如

```
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7-m/nofp/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7-a+simd/hard/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7e-m+dp/softfp/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7-a+fp/hard/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7-a+simd/softfp/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7+fp/hard/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7-r+fp.sp/softfp/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v8.1-m.main+mve/hard/libgcc.a
gcc-arm-none-eabi: /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v7e-m+dp/hard/libgcc.a
```

xry111, 2024-04-09:
按道理来说应该用 `-mfpu=none` 而不是 `-mno-lsx`，因为你光禁掉 LSX 的话并不禁止浮点指令，只是目前我们并不用浮点指令优化整数操作才碰巧不出问题。一个很显然的优化是拿浮点寄存器作为定点寄存器的 spill 空间，这样可以避免往栈上 spill，只是现在没人写这优化罢了。

用了 `-mfpu=none` 以后它会把 ABI 降级成 `-mabi=lp64s`，所以会用 LP64S 的 multilib，这样可以在编译 GCC 的时候把 LP64S multilib 加上 `-mfpu=none` (`--with-multilib-list=lp64d,lp64s/-mfpu=none`)。

但是这意味着为了编译 bootloader 而让用户空间里面的所有 LP64S ABI 代码一起变慢，我很讨厌这种为了一个包坑其他所有包的做法。

而且还有一个问题是 ABI 降级以后甚至需要一个 LP64S 的 Glibc (提供 limits.h)，我就编译个 bootloader 居然需要一个 Glibc？bootloader 逻辑上就不能用 Glibc 啊？！

反正我选择用 GRUB。

MarsDoge, 2025-04-29:
Can this issue be closed? EFI environment does not support SIMD instructions.

closed 2025-04-29 by MarsDoge