loongson/Firmware #99: 求魔方 M700S 固件

Qinka, 2024-04-28:
话说，有魔方 M700S 相关的固件么？（原谅伸手党直接提问）。

lang-H, 2024-07-08:
同求NUC的新固件，希望可以把第二个网口用上

olljanat, 2025-01-23:
Which firmware version you have now?

My Morefine M700S had this firmware installed in factory:
- BIOS Version: Loongson-UDK2018-V4.0.05759-stable202405
- Computer Type: Loongson-3A6000-7A2000-NUC

lang-H, 2025-01-24:
应该是可以直接用中科云nuc的固件：
https://bbs.ctcisz.com/forum.php?mod=viewthread&tid=115&extra=page%3D1

olljanat, 2025-01-25:
It definitely looks same hardware with just different branding. I can try that after my BIOS burner arrives and will be able to backup factory firmware first.

However, can someone create pull request which would upload it here? It looks to be impossible for me to get registered to Baidu.

PR #134 (2025-02-10): NUC_QA612N0: Base LsRefCodePkg V4.0.05823-stable202408
Closes #99