loongson/Firmware #112: 请求增加华硕龙芯主板BIOS

haowuge, 2024-08-18:
目前华硕主板没有官方更新渠道，请求增加华硕龙芯主板BIOS

xen0n, 2024-08-18:
FYI: #111 开了又关了，没有说明情况

haowuge, 2024-08-18:
可能有bug吧，看内存频率XD
我测试过，关闭的固件跟0804，风扇在启动的时候会明显更吵，跑大负载也感觉也更吵，0206的内置固件是没特殊反应（猜可能那时候没这功能？）

[image]
[image]

> FYI: #111 开了又关了，没有说明情况

xry111, 2024-08-18:
> 可能有bug吧，看内存频率XD

人类感谢你 (doge)

dg1vg4, 2024-09-04:
https://bbs.loongarch.org/d/457-3a6000xc-ls3a6m-abios20240804

haowuge, 2024-09-04:
> https://bbs.loongarch.org/d/457-3a6000xc-ls3a6m-abios20240804

那个非官方性质，没任何其他信息（比如更新日志，哈希值等等）

moontide, 2024-09-16:
还是关了吧，发在这里不合适，就好像你买了店家 A 的商品却去找店家 B 质保一样…

华硕主板的 BIOS 肯定在华硕那边找，就算是龙芯跟华硕是合作伙伴，龙芯这边最多也只能加个友情链接…

driver1998, 2026-01-13:
华硕的固件支持页上线了，不过正常从首页搜索还是搜不到这个板子：
https://www.asus.com.cn/supportonly/xc-ls3a6m/helpdesk_bios/

xen0n, 2026-01-13:
> > FYI: [#111](https://github.com/loongson/Firmware/pull/111) 开了又关了，没有说明情况
> 
> 现在应该可以提及了，当时其实是asus不让放，被通知了，asus 后续请关注官方支持页吧.

可以建一个文件夹但在它的 README 里给个官方支持页面链接，导流到 ASUS？

closed 2026-11-02 by kilaterlee