loongson/Firmware #115: XC LS3A6M with RX6400 cannot enter BIOS, let alone the OS

RichardBrowning, 2024-09-03:
标题已经讲明白了，请问有什么已知的解决方法吗？似乎现在RDNA1以上的显卡都没办法正常显示。RX550可以。

问题概述：
运行LoongarchLinux的华硕龙芯3A6000主板，用radeon RX 6400显卡，无论<输出>选择“自动”还是“外接显卡（就是指dGPU）”，bios和系统均无显示<-有信号输入但是完全黑屏

整体配置：
操作系统：LoongArchLinux （内核版本6.8）
主板：华硕 xc-ls3a6m（没有A）
BIOS版本：0804
内存条：单条16GB紫光国芯DDR4 3200
硬盘：长城某型号256G SATA硬盘
显卡：AMD Radeon RX6400 4G / Radeon RX550 4G
网卡：英特尔AX-200

问题产生方法：
把RX6400插在PCIe x16插槽上，BIOS设置中把Resizable BAR和Above 4G打开，输出设置为自动或外接显卡，开机就会触发

已知尝试：
修改启动参数，添加loongson.setmode=0等，无效

dg1vg4, 2024-09-04:
亮固件这事，据说RDNA全系都有支持问题。
只能回去用GCN架构卡。

xry111, 2024-09-04:
但是这个应该不影响进系统以后显示，可以用 ssh 或者串口采集一个内核 log 以供分析吗？

dg1vg4, 2024-09-04:
> 但是这个应该不影响进系统以后显示，可以用 ssh 或者串口采集一个内核 log 以供分析吗？

很遗憾，个人的龙芯电脑都是插GCN架构卡的。

xry111, 2024-09-04:
> > 但是这个应该不影响进系统以后显示，可以用 ssh 或者串口采集一个内核 log 以供分析吗？
> 
> 很遗憾，个人的龙芯电脑都是插GCN架构卡的。

我问楼主 :)

RichardBrowning, 2024-09-04:
> 但是这个应该不影响进系统以后显示，可以用 ssh 或者串口采集一个内核 log 以供分析吗？

我最近在OSU的学校宿舍，他们用的PEAP的无线网，实在没法无线。有线正在尝试，不过因为内核没有崩溃，其实journalctl里面能保存下来log，其内容正在研究

dg1vg4, 2024-11-29:
倒是有一个可能可行的亮BIOS解决办法，那就是改刷百敖的一个BIOS，但是那个BIOS版本是针对龙芯天创者XA61200开发的，不知道能不能在华硕的主板上使用。