loongson/Firmware #116: 希望能找到联想开天M540z的BIOS固件更新

dg1vg4, 2024-09-04:
联想KaiTianM540Z-D024

# dmidecode 3.2

Getting SMBIOS data from sysfs.

SMBIOS 3.2.0 present.

Table at 0x90000000FE824000.

Handle 0x0000, DMI type 0, 24 bytes

BIOS Information

Vendor: ZD-TECH

Version: W05KT18A

Release Date: 09/24/2021

ROM Size: 8192 kB

Characteristics:

	PCI is supported

	BIOS is upgradeable

	BIOS shadowing is allowed

	Boot from CD is supported

	Selectable boot is supported

	BIOS ROM is socketed

	Serial services are supported (int 14h)

	USB legacy is supported

	Function key-initiated network boot is supported

	UEFI is supported

BIOS Revision: 4.0

System Information

Manufacturer: KaiTianM540Z-D024

Product Name: 90A1A006KX

Version: Lenovo

Serial Number: KT006QN8

UUID: 94f493ee-83bf-4f8d-b1a4-2a55098e25fd

Wake-up Type: Power Switch

SKU Number: Loongson_SKU

Family: Type1Family
Handle 0x0002, DMI type 2, 15 bytes

Base Board Information

Manufacturer: Loongson

Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB

Version: 1.0

Serial Number: MX20BA001J

Asset Tag: Type2 - Board Asset Tag

Features:

	Board is a hosting board

	Board is replaceable

Location In Chassis: Type2 - Board Chassis Location

Chassis Handle: 0xFFFF

Type: Motherboard

Contained Object Handles: 0
Handle 0x0003, DMI type 3, 22 bytes

elysia-best, 2025-07-02:
我的主板是L5LMB01，使用的BIOS是这个

[Loongbios_LXKT_L5LMB01.zip](https://github.com/user-attachments/files/21021870/Loongbios_LXKT_L5LMB01.zip)

BIOS 版本：Loongson-UDK2018-U2.0.04011-beta2
显示的主板型号：Loongson-LS3A5000-7A1000-1w-v0.1-CRB 

[image]

我目前也只有2021年的版本，希望对其他人有用
```
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Table at 0x90000000FD083000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: Loongson
        Version: Loongson-UDK2018-V2.0.04011-beta2
        Release Date: 03/10/2021
        ROM Size: 4 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                Serial services are supported (int 14h)
                USB legacy is supported
                UEFI is supported
        BIOS Revision: 2.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Loongson
        Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
        Version: Not Specified
        Serial Number: Not Specified
        UUID: Not Present
        Wake-up Type: Power Switch
        SKU Number: Not Specified
        Family: Not Specified

Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
        Manufacturer: Loongson
        Product Name: Loongson-LS3A5000-7A1000-1w-EVB-V1.21
        Version: Not Specified
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Not Specified
        Chassis Handle: 0x0000
        Type: Motherboard
        Contained Object Handles: 0

Handle 0x0003, DMI type 3, 24 bytes
Chassis Information
        Manufacturer: Loongson
        Type: Desktop
        Lock: Not Present
        Version: Not Specified
        Serial Number: Not Specified
        Asset Tag: Not Specified
        Boot-up State: Safe
        Power Supply State: Safe
        Thermal State: Safe
[... 162 lines cut ...]

Handle 0x0010, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected

Handle 0x0011, DMI type 39, 22 bytes
System Power Supply
        Power Unit Group: 1
        Location: To Be Filled By O.E.M
        Name: To Be Filled By O.E.M
        Manufacturer: To Be Filled By O.E.M
        Serial Number: To Be Filled By O.E.M
        Asset Tag: To Be Filled By O.E.M
        Model Part Number: To Be Filled By O.E.M
        Revision: To Be Filled By O.E.M
        Max Power Capacity: Unknown
        Status: Present, Unknown
        Type: Unknown
        Input Voltage Range Switching: Unknown
        Plugged: No
        Hot Replaceable: Yes

Handle 0xFEFF, DMI type 127, 4 bytes
End Of Table
```

elysia-best, 2026-09-29:
这是2023年4月的最新固件：
```
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Table at 0x90000000FD68C000.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: ZD-TECH
        Version: W05KT20A
        Release Date: 04/26/2023
        ROM Size: 8 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                Serial services are supported (int 14h)
                USB legacy is supported
                Function key-initiated network boot is supported
                UEFI is supported
        BIOS Revision: 4.0

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
        Version: 1.0
        Serial Number: TBD by OEM
        UUID: 00112233-4455-6677-8899-aabbccddeeff
        Wake-up Type: Power Switch
        SKU Number: Loongson_SKU
        Family: Type1Family

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: Loongson
        Product Name: Loongson-LS3A5000-7A1000-1w-V0.1-CRB
        Version: 1.0
        Serial Number: MX20BA001J
        Asset Tag: Type2 - Board Asset Tag
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Type2 - Board Chassis Location
        Chassis Handle: 0xFFFF
        Type: Motherboard
        Contained Object Handles: 0
```

[W05KT20A.zip](https://github.com/user-attachments/files/32809440/W05KT20A.zip)