loongson/Firmware #117: System occasionally fails to boot after software reset.

shankerwangmiao, 2024-09-04:
- Board: `XA61200_V1_1`
- Firmware: `UDK2018-3A6000-7A2000_EVB_V4.0.05756_prestable2405_0523dbg.fd`

Boot log:

```
[   96.850444] systemd-shutdown[1]: Rebooting.
[   97.041644] ACPI: PM: Preparing to enter system sleep state S5
[   97.080884] kvm: exiting hardware virtualization
[   97.118064] reboot: Restarting system
ResetType: EfiResetCold.
Shut down slave cores done!

UEFI-EDK2 loongson Initializing...

POSTCODE=<0x00>

CPU Type: 3A6000-HV
CPU Version: 00000000
CPU Level: 00000000
CPU SRAM: 00000000
Node 00000000
N Voltage  write :
v ctrl end

N Voltage  read :
000000b4

P Voltage write :
io ctrl end
AVS: Get Vddn value is: 0000047a
AVS: Get Vddp value is: 00000418

(0x1fe00190)  : 0000428200000000
CPU CLK SEL : 00000002
MEM CLK SEL : 00000014
HT CLK SEL : 00000010

POSTCODE=<0x01>

POSTCODE=<0x02>
Disable HT0 clock
Change the scale of LS132 clock
Soft CLK SEL adjust begin
CORE & NODE:08110f85
HT :00000000Loongson3Ht1Addr32Trans End

POSTCODE=<0x06>
Copy Sec code to SCache-As-Ram.
Copy PEI code to SCache-As-Ram.

POSTCODE=<0x07>
Entering C environment
POSTCODE=<0x08>
&SecCoreData.DataSize=9027FFA0 SecCoreData.DataSize=48
&SecCoreData.TemporaryRamSize=9027FFC0 SecCoreData.TemporaryRamSize=300000
&SecCoreData.TemporaryRamBase=9027FFB8 SecCoreData.TemporaryRamBase=90100000
&SecCoreData.PeiTemporaryRamBase=9027FFC8 SecCoreData.PeiTemporaryRamBase=90100000
&SecCoreData.PeiTemporaryRamSize=9027FFD0 SecCoreData.PeiTemporaryRamSize=100000
&SecCoreData.StackBase=9027FFD8 SecCoreData.StackBase=90200000
&SecCoreData.StackSize=9027FFE0 SecCoreData.StackSize=200000
&SecCoreData.BootFirmwareVolumeBase=9027FFA8 SecCoreData.BootFirmwareVolumeBase=90020000
&SecCoreData.BootFirmwareVolumeSize=9027FFB0 SecCoreData.BootFirmwareVolumeSize=E0000
POSTCODE=<0x09>
Find Pei EntryPoint=90020360
Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
Install PPI: 8C8CE578-8A3D-4F1C-9935-896185C32DD3
Install PPI: 5473C07A-3DCB-4DCA-BD6F-1E9689E7349A
The 0th FV start address is 0x00090020000, size is 0x000E0000, handle is 0x90020000
Register PPI Notify: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
[... 2079 lines cut ...]
POSTCODE=<0x10> PROGRESSCODE=<V03020002>
Install PPI: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
Notify: PPI Guid: 49EDB1C1-BF21-4761-BB12-EB0031AABB39, Peim notify entry point: 90026BA8
The 5th FV start address is 0x000C00C0000, size is 0x00260000, handle is 0xC00C0000
Customized Guided section Memory Size required is 0xD9C010 and address is 0xFE100000
Extract guided section Failed - Invalid Parameter
Found firmware volume Image File C00C0048 in FV[5] C00C0000
Customized Guided section Memory Size required is 0xD9C010 and address is 0xFD353000
Extract guided section Failed - Invalid Parameter
Customized Guided section Memory Size required is 0x81010 and address is 0xFD2C1000
Found firmware volume Image File C02DAC78 in FV[5] C00C0000
ProcessFvFile() FV at 0xFD2C1010, FvAlignment required is 0x10
Install PPI: EA7CA24B-DED5-4DAD-A389-BF827E8F9B38
Notify: PPI Guid: EA7CA24B-DED5-4DAD-A389-BF827E8F9B38, Peim notify entry point: 90026BA8
The 3th FV start address is 0x000FD2C1010, size is 0x00081000, handle is 0xFD2C1010
Install PPI: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
Notify: PPI Guid: 49EDB1C1-BF21-4761-BB12-EB0031AABB39, Peim notify entry point: 90026BA8
The Fv FD2C1010 has already been processed!
POSTCODE=<0x11> PROGRESSCODE=<V03020003>
FV file C02DAC78 has been dispatched!
DXE IPL Entry

ASSERT_EFI_ERROR (Status = Not Found)
ASSERT [DxeIpl] /data/qiandongyan/push/zuixin/push2/rd/uefi-loongarch-clean-update/MdeModulePkg/Core/DxeIplPeim/DxeLoad.c(469): !EFI_ERROR (Status)
```

MarsDoge, 2025-03-13:
There are two situations where this problem may occur. 
One is that the flash model or voltage does not match the instruction fetch error, and the other is that the memory is unstable. @shankerwangmiao

MarsDoge, 2025-03-13:
Is this a probability issue or a certainty issue?
Maybe it is also related to the second case: getting data from the flash to the memory in the early stage.

shankerwangmiao, 2025-03-13:
This issue occurs frequently but not always, since the system can finally boot. The frequency is higher when the system reboots than during cold start