loongson/Firmware #118: XA612A0 固件prestable2405rel.fd 大小和stable2311pllrel.fd不一致

Leonids1993, 2024-09-14:
2405的bios比老版大4M，是否有问题？老版4M固件是否有更新？
``` shell
david@DESKTOP-EA9AHBO:~/Firmware/6000Series/PC/XA612A0$ ls -alh
total 24M
drwxr-xr-x 2 david david 4.0K Sep 14 10:33 .
drwxr-xr-x 4 david david 4.0K Sep 14 10:33 ..
-rw-r--r-- 1 david david  454 Sep 14 10:33 README.md
-rw-r--r-- 1 david david  464 Sep 14 10:33 SHA256SUMS.txt
-rw-r--r-- 1 david david 8.0M Sep 14 10:33 UDK2018-3A6000-7A2000_XA612A0_prestable2405dbg.fd
-rw-r--r-- 1 david david 8.0M Sep 14 10:33 UDK2018-3A6000-7A2000_XA612A0_prestable2405rel.fd
-rw-r--r-- 1 david david 4.0M Sep 14 10:33 UDK2018-3A6000-7A2000_XA612A0_stable2311plldbg.fd
-rw-r--r-- 1 david david 4.0M Sep 14 10:33 UDK2018-3A6000-7A2000_XA612A0_stable2311pllrel.fd
```

MarsDoge, 2024-09-14:
> 2405的bios比老版大4M，是否有问题？老版4M固件是否有更新？
> 
8M文件中后4M为功能扩展，可只烧录前4M数据作为基本功能使用或者如果FLash支持8M亦可烧录完整功能.

moontide, 2024-09-16:
似乎加了新功能，我的  XA61200 的新固件大小也是 8M，里面加了调频的功能。

<pre>
262129 003fff00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262130 003fff10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262131 003fff20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262132 003fff30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262133 003fff40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262134 003fff50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262135 003fff60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262136 003fff70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262137 003fff80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262138 003fff90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262139 003fffa0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262140 003fffb0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262141 003fffc0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262142 003fffd0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262143 003fffe0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262144 003ffff0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff  ................
262145 00400000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
262146 00400010: 78 e5 8c 8c 3d 8a 1c 4f 99 35 89 61 85 c3 2d d3  x...=..O.5.a..-.
262147 00400020: 00 00 20 00 00 00 00 00 5f 46 56 48 ff fe 04 00  .. ....._FVH....
262148 00400030: 48 00 af e4 00 00 00 02 00 02 00 00 00 10 00 00  H...............
262149 00400040: 00 00 00 00 00 00 00 00 52 ef 1c 9c d1 eb d7 48  ........R......H
262150 00400050: be 7f 5d d9 e1 da cc 11 5d aa 0b 00 56 61 02 f8  ..].....]...Va..
262151 00400060: 3e 61 02 02 98 58 4e ee 14 39 59 42 9d 6e dc 7b  >a...XN..9YB.n.{
262152 00400070: d7 94 03 cf 18 00 01 00 5d 00 00 00 01 10 b0 0b  ........].......
262153 00400080: 00 00 00 00 00 00 06 00 30 cd 48 b5 45 a1 07 95  ........0.H.E...
262154 00400090: b9 8f ce ae 60 03 c5 cb 26 d4 e3 3a 6d a0 4f 8f  ....`...&..:m.O.
262155 004000a0: eb 39 da a1 ca 17 31 38 fd 24 6a a0 68 ed e3 95  .9....18.$j.h...
262156 004000b0: 32 1a 0c c0 7e 97 f2 8e eb 16 13 22 59 5d a7 67  2...~......"Y].g
262157 004000c0: b7 9d a5 47 f9 07 70 5d e1 92 42 40 51 3b 99 e1  ...G..p]..B@Q;..
262158 004000d0: d3 c2 d4 67 62 7b ab 0f 1a 2c 7a ab 0e 06 e3 21  ...gb{...,z....!
262159 004000e0: e6 ca a2 6a 92 ec aa c5 55 67 75 f7 bf 2b 25 29  ...j....Ugu..+%)
262160 004000f0: a0 e7 ef 34 da 2b 13 b8 46 ad 04 63 6d 28 e0 86  ...4.+..F..cm(..
262161 00400100: 00 9d f0 cc c3 ab 1e 4e a0 53 72 1a 8b be b0 80  .......N.Sr.....
</pre>

[image]

closed 2024-09-23 by Leonids1993