loongson/Firmware #119: Compatibility with SN550 (and maybe other WD NVME disks)

xry111, 2024-10-06:
With SN550 on XA61200, Linux fail to mount a partition on the disk R/W with:

```
nvme0n1: I/O Cmd(0x1) 0 LBA 411640, 0 blocks, I/O Error (sct 0x0 / sc 0x4) MORE
I/O Error, dev nvme0n1, sector 411640 op 0x1:(WRITE) flags 0x20000 phys_reg 1 prio class 0
Buffer I/O error on dev nvme0n1p3, logical block 0, lost sync page write
```

Someone told me this affects other motherboard models and other WD disk models as well, and it can be fixed (or worked around) by a firmware update.  But the update isn't released here as at now.

dg1vg4, 2024-11-29:
我问一下，这个固态硬盘，是否曾经在Windows平台上使用过？
而且在使用的时候，没有关闭快速启动？
如果是，那可能是Windows的一套安全机制锁死了该分区的读写，需要在Windows系统下，进入控制面板-电源选项-选择电源按钮页面，如图所示，关闭“快速启动”
[image]

xry111, 2024-11-30:
> 我问一下，这个固态硬盘，是否曾经在Windows平台上使用过？ 而且在使用的时候，没有关闭快速启动？ 如果是，那可能是Windows的一套安全机制锁死了该分区的读写，需要在Windows系统下，进入控制面板-电源选项-选择电源按钮页面，如图所示，关闭“快速启动” ![image](https://private-user-images.githubusercontent.com/28698914/391137127-c69c84dc-b231-4b0e-9ef9-eb26ce86b85d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI5NTQwMTIsIm5iZiI6MTczMjk1MzcxMiwicGF0aCI6Ii8yODY5ODkxNC8zOTExMzcxMjctYzY5Yzg0ZGMtYjIzMS00YjBlLTllZjktZWIyNmNlODZiODVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTMwVDA4MDE1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5NDk5Y2FjNTEyZTM4MTAxMjc4NmIyZmI4N2JkMjUzMTJjY2U0MGZiMmZhNzIxMTY1Y2Y1OTMwNTlkODk0NmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7CYcvQE95vFd7k58z874TNs_FeJ_f8Xp6XF8wNneBV8) 如果还是不想折腾，干脆全盘格式化，然后换EXT4分区得了。

已经是一个 ext4 分区了，而且插 x86 机器里面没问题。