loongson/Firmware #127: MultiArchUefiSupport 固件无法使用 NVIDIA RTX2070S 显卡

KatyushaScarlet, 2024-12-08:
* 显卡型号
  * 影驰 NVIDIA RTX 2070 S

* 已测试主板：
  * 3A5000-7A2000-EVB
  * XA61200

* 具体表现
  * 3A5000-7A2000-EVB
    * 无论是板载HDMI输出还是显卡HDMI输出，均黑屏
  * XA61200
    * 只有板载HDMI有输出，但固件内PCIe信息找不到显卡，系统中`lspci`命令找不到显卡

kilaterlee, 2024-12-09:
这里面有两个问题：
1. 3A5000平台是否存在不启动问题？根据你的描述，是不插卡也会黑屏无显是这样吗？
2. `lspci`找不到的问题需要具体排查，如果方便可以将出问题的卡发给我们看一下。

KatyushaScarlet, 2024-12-09:
> 这里面有两个问题：
> 
> 1. 3A5000平台是否存在不启动问题？根据你的描述，是不插卡也会黑屏无显是这样吗？
> 2. `lspci`找不到的问题需要具体排查，如果方便可以将出问题的卡发给我们看一下。

1. 不存在。不插这张独显或者插AMD GCN显卡均可正常使用。
2. 可以，您可通过我的Github用户名添加微信联系我。

kilaterlee, 2024-12-09:
> > 这里面有两个问题：
> > 
> > 1. 3A5000平台是否存在不启动问题？根据你的描述，是不插卡也会黑屏无显是这样吗？
> > 2. `lspci`找不到的问题需要具体排查，如果方便可以将出问题的卡发给我们看一下。
> 
> 1. 不存在。不插这张独显或者插AMD GCN显卡均可正常使用。
> 2. 可以。我通过邮件联系你吧

已经给您outlook邮箱发信了，如果看到请回复

KatyushaScarlet, 2024-12-09:
> > > 这里面有两个问题：
> > > 
> > > 1. 3A5000平台是否存在不启动问题？根据你的描述，是不插卡也会黑屏无显是这样吗？
> > > 2. `lspci`找不到的问题需要具体排查，如果方便可以将出问题的卡发给我们看一下。
> > 
> > 
> > 
> > 1. 不存在。不插这张独显或者插AMD GCN显卡均可正常使用。
> > 2. 可以。我通过邮件联系你吧
> 
> 已经给您outlook邮箱发信了，如果看到请回复

您好，我的邮箱无法送达到您的邮箱。也可以用我的Github用户名添加我的微信联络

kilaterlee, 2024-12-09:
> > > > 这里面有两个问题：
> > > > 
> > > > 1. 3A5000平台是否存在不启动问题？根据你的描述，是不插卡也会黑屏无显是这样吗？
> > > > 2. `lspci`找不到的问题需要具体排查，如果方便可以将出问题的卡发给我们看一下。
> > > 
> > > 
> > > 
> > > 1. 不存在。不插这张独显或者插AMD GCN显卡均可正常使用。
> > > 2. 可以。我通过邮件联系你吧
> > 
> > 
> > 已经给您outlook邮箱发信了，如果看到请回复
> 
> 您好，我的邮箱无法送达到您的邮箱。也可以用我的Github用户名添加我的微信联络

已经发送添加请求

closed 2024-12-18 by KatyushaScarlet

xen0n, 2024-12-18:
刚收到 issue closed 的推送, 结论是啥？方便的话可以同步一下

KatyushaScarlet, 2024-12-18:
> 刚收到 issue closed 的推送, 结论是啥？方便的话可以同步一下

有点玄学，XA612A0更换了内存与显示器，在固件界面点亮了。
之前点不亮的环境没有进一步研究，3A5000 EVB也没有继续研究。
因为今天看到了问题汇总的issue，打算过几天再测，如果有新信息会发到那个issue下