loongson/Firmware #128: Collecting Validation and Usage Reports for Various GPU Series in UEFI

MarsDoge, 2024-12-18:
## Description
To better understand the compatibility and performance of our UEFI project across different GPUs, we encourage developers to share their testing results and usage experiences. Your feedback is crucial for improving the project!  

## Submission Requirements
Please provide your GPU validation and usage information using the template below.  

### Test Environment
- **GPU Brand and Model**: (e.g., NVIDIA RTX 4090 or AMD Radeon RX 6900 XT)  
- **GPU Driver Version**: (if applicable, e.g., UEFI GOP version or driver firmware version)  
- **UEFI Firmware and Version**: (e.g., edk2_202408_LoongArch64_pc_XA61200_MultiArchSupport_rel.fd)  
- **Testing Platform**: (e.g., 3A6000 XA61200, etc.)  

### Test Results
- **Did it run successfully?**: Yes
- **Feature Testing**:  
  - UEFI GOP support: Passed
  - Display output: Passed
  - Resolution support: (e.g., 1920x1080 supported, etc.)  
  - Performance: (e.g., X11perf, 3D Mark Test Record)  
- **Bugs or Issues Found**: (If any, include steps to reproduce, logs, and other relevant details)  

### Additional Information
- **Were specific UEFI Setup enabled or tested?**: (e.g., Enable Above4G, ResizeBar, Sriov, etc.)  
- **Other notes or observations**:  Nvidia 4090 operating system layer has poor performance using Nouveau driver.

---

We look forward to your feedback! If you have any questions, feel free to discuss them in this issue.
With this, let's build up our GPUs support list.

MultiArchUefiSupport UEFI Binary: https://github.com/loongson/Firmware/tree/main/MultiArchUefiSupport

xen0n, 2024-12-18:
Template for easy copy-pasting:

```markdown
### Test Environment

* GPU Brand and Model: xx
* GPU Driver Version
    * UEFI GOP version: xx / I don't know
    * GPU firmware version: xx / I don't know
* UEFI Firmware and Version: `edk2_202408_LoongArch64_pc_XA61200_MultiArchSupport_rel.fd`
* Testing Platform: your motherboard model

### Test Results

* Did it run successfully?: Yes / No
* Feature Testing
    * UEFI GOP support: works / FAILS: symptoms
    * Display output: works / FAILS: symptoms
    * Resolution support: e.g., 1920x1080 supported, etc.
    * Performance
        * x11perf: not tested / XXX
        * 3D Mark Test Record: not tested / XXX
        * other benchmarks you have run
* Bugs or Issues Found
    * if any: describe in items
    * if none: n/a

### Additional Information

* Were specific UEFI Setup enabled or tested?
    * Above4G: not applicable / not tested / works / FAILS: symptoms
    * Resizable BAR: not applicable / not tested / works / FAILS: symptoms
    * SR-IOV: not applicable / not tested / works / FAILS: symptoms
* Other notes or observations
    * if any: describe with items
    * if none: n/a
```

方便复制粘贴的中文模板:

```markdown
### 测试环境

* GPU 品牌与型号: xx
* GPU 驱动版本
    * UEFI GOP 版本: xx / 不知道
    * GPU 固件版本: xx / 不知道
* UEFI 固件版本: `edk2_202408_LoongArch64_pc_XA61200_MultiArchSupport_rel.fd`
* 测试平台: 您的主板型号

### 测试结果

* 正常工作吗? 是 / 否
* 功能测试
    * UEFI GOP 支持: works / FAILS: symptoms
    * 显示输出: works / FAILS: symptoms
    * 分辨率支持情况: e.g., 1920x1080 supported, etc.
    * 性能
        * x11perf: 未测试 / XXX
        * 3D Mark 跑分: 未测试 / XXX
        * 其他您测试过的项目
* 发现的问题
    * 如有: 逐条描述
    * 如无: 无

### 额外信息

* 是否开启其他 UEFI 特性测试过?
    * Above4G: 不适用 / 未测试 / 工作正常 / 工作不正常: 症状
    * Resizable BAR: 不适用 / 未测试 / 工作正常 / 工作不正常: 症状
    * SR-IOV: 不适用 / 未测试 / 工作正常 / 工作不正常: 症状
* 其他观察结果与笔记
    * 如有: 逐条描述
    * 如无: 无
```

xry111, 2024-12-18:
> * **Other notes or observations**:  Nvidia 4090 operating system layer has poor performance using Nouveau driver.

Just FYI: on x86 it's reported using Zink on the top of the Nouveau Vulkan driver produces a better performance than the Nouveau Gallium driver.  Maybe it can help LoongArch too.

dtutu0203, 2025-02-22:
### 测试环境

* GPU 品牌与型号: AMD7900
* GPU 驱动版本
    * UEFI GOP 版本: xx / 不知道
    * GPU 固件版本: xx / 不知道
* UEFI 固件版本: EFI_3A6000-XA61200_Emu_1128_v2_dbg.fd/EFI_3A6000-XA61200_Emu_1128_v2_ref.fd
* 测试平台: 您的主板型号 XA61200_v1_1

### 测试结果

* 正常工作吗? 是 
* 功能测试
    * UEFI GOP 支持: 工作正常 / 工作不正常: 症状
    * 显示输出: 工作正常 / 工作不正常: 症状
    * 分辨率支持情况: 例: 支持 4K、2K等等
    * 性能
        * x11perf: 未测试
        * 3D Mark 跑分: 未测试
        * 其他您测试过的项目
* 发现的问题
    * 如有: 逐条描述
    * 如无: 无

### 额外信息

* 是否开启其他 UEFI 特性测试过?
    * Above4G: 工作正常
    * Resizable BAR: 工作正常
    * SR-IOV:  未测试 
* 其他观察结果与笔记
    * 如有: 逐条描述
    * 如无: 无

dtutu0203, 2025-02-22:
### 测试环境

* GPU 品牌与型号: xx
* GPU 驱动版本
    * UEFI GOP 版本: xx / 不知道
    * GPU 固件版本: xx / 不知道
* UEFI 固件版本: `edk2_202408_LoongArch64_pc_XA61200_MultiArchSupport_rel.fd` 
* 测试平台: 您的主板型号

### 测试结果

* 正常工作吗? 是 
* 功能测试
    * UEFI GOP 支持: 工作正常
    * 显示输出: 工作正常
    * 分辨率支持情况:  支持 4K、2K等等
    * 性能
        * x11perf: 未测试
        * 3D Mark 跑分: 未测试
        * 其他您测试过的项目
* 发现的问题
    * 如有: 逐条描述
    * 如无: 无

### 额外信息

* 是否开启其他 UEFI 特性测试过?
    * Above4G: 工作正常 
    * Resizable BAR: 工作正常
    * SR-IOV:  未测试
* 其他观察结果与笔记
    * 如有: 逐条描述
    * 如无: 无

FanFansfan, 2025-03-22:
### 测试环境

* GPU 品牌与型号: Intel DG1
* GPU 驱动版本
    * UEFI GOP 版本: 不知道
    * GPU 固件版本: 不知道
* UEFI 固件版本: `edk2_202408_LoongArch64_pc_XA61200_MultiArchSupport_rel.fd`
* 测试平台: XA61200

### 测试结果

* 正常工作吗? 是
* 功能测试
    * UEFI GOP 支持:工作不正常: 不显示
    * 显示输出: 工作正常
    * 分辨率支持情况: 4k
    * 性能
        * x11perf: 未测试
        * 3D Mark 跑分: 未测试
* 发现的问题
    * 每次刷完固件必须拔卡设置Above4G。

### 额外信息

* 是否开启其他 UEFI 特性测试过?
    * Above4G: 工作正常
    * Resizable BAR: 不适用
    * SR-IOV: 不适用
* 其他观察结果与笔记
    * 无